exelban/stats #1458: Fan Controlling is not working anymore for me since update

vrjo, 2023-05-08:
Amazing piece of software, could have come from apple themselves. Very sexy. 

Im on a MBP 2021 with M1 Pro and 2 weeks ago I could control the fans via the tool but not anymore. It's just not reacting to any command. Before it worked. Problem is not occurring since the last update but since the update before the last update.

[image]

Jerry23011, 2023-05-08:
You might want to try uninstalling the fan control helper in Settings and reinstall it.

vrjo, 2023-05-08:
Amazing, the reinstall of the fan control helper worked! FIXED. Thank you for your quick response.

[image]

Jerry23011, 2023-05-08:
Great to hear it's fixed!

closed 2023-05-08 by exelban