exelban/stats #1461: Show gauge on collapsed fan view

LostInCompilation, 2023-05-09:
Now with the new release we have the ability to hide the "Automatic/Manual" controls for fans in the sensor popup. A really nice design feature would be to add a gauge for each fan, showing the current fan percentage. 

Right now collapsed view looks like this:
[image]

My suggestion would be something like this:
[image]

Also maybe making the gauges clickable to change the fan speed in collapsed mode. But I don't think that would be so easy to add, so just a static display like the bar for used disk space would be cool.

exelban, 2023-05-09:
Hi. Thanks. But it most probably will not be added. At least not in this way.
First of all, it's not readable at all for me. I will prefer a light bottom border that indicates that.
The second reason, after I add this functionality in a few days someone come and ask the same for sensors.

closed 2023-05-28 by exelban

LostInCompilation, 2023-05-31:
@osy Looks good. However I somehow find it a bit out of design, that the progress bar background is white. Every element in Stats has a somehow dark gray color (when in dark mode). The progress bar background of the fans is the only white thing. When the fans are completely off, just two white rectangles appear for the progress bar. 

Would it be possible to change the color of them?

osy, 2023-05-31:
I’m sorry, what’s the context here?

LostInCompilation, 2023-05-31:
@osy Sorry that was a mistake 😅 Mentioned the wrong person.