exelban/stats #1462: Fan control is not working

GerritHanno, 2023-05-09:
Hey there,

Fan control doesn't work anymore. Anyone else with this problem? (Macbook Pro 13" 2019 -  2 Thunderbolt Ports).
Other programs like "Macs Fan Control" can still adress the Fan and control it.

Restart, Reset SMC, Reinstall, nothing fixed it

exelban, 2023-05-09:
hi. Share please output of `launchctl print-disabled system `.

Whiteangel1993, 2023-05-09:
disabled services = {
		"com.apple.CSCSupportd" => disabled
		"com.apple.ftpd" => disabled
		"com.apple.mdmclient.daemon.runatboot" => disabled
		"com.google.keystone.daemon" => enabled
		"com.microsoft.autoupdate.helper" => enabled
	}

Whiteangel1993, 2023-05-09:
thx. does'nt look like:

3029   0,0  0,0 33587684    252 s000  R+    6:22pm   0:00.01 grep SMC

exelban, 2023-05-09:
nothing else?

und3fined, 2023-05-09:
Same with me

exelban, 2023-05-09:
strange. Could you try to install -> uninstall the helper?

Whiteangel1993, 2023-05-09:
Got it.
1. I uninstalled Fancontrol in Stats
2. Couldn't reinstall it -> Strange!
3.  Deleted Stats completely with "AppCleaner" and reinstalled it
4. Installed Fancontrol again
5. It works!

exelban, 2023-05-09:
strange. Unfortunately it's not a first time. And I cannot reproduce that bug. And have no idea why it happens(

Whiteangel1993, 2023-05-26:
I finally found a better way to fix this issue:

1. Uninstall "Fan Control" in the Stats - Settings
2. Close Stats
3. (Donwload/Install and then) Open the App "AppCleaner"
4. In the right upper corner choose the "list"-symbol
5. Search and click on "Stats"
6. Mark every document and folder EXCEPT Stats (/Application/Stats.app)
7. Click delete
8. Open Stats (Your settings should not be removed)
9. Reinstall Fan Control in Stats
10. Voila!

Ryionaw, 2023-07-07:
help please i cant turn on the fan after installing

IceDev-a, 2023-07-15:
Re-install only the fan helper solved the problem temporary. It won't work next time you restart your mac. To solve the problem, you need to reinstall the Stats.

afonsodemello, 2024-01-15:
> strange. Unfortunately it's not a first time. And I cannot reproduce that bug. And have no idea why it happens(

I also had that bug and fixed it, I just uninstalled completely stats with appcleaner then reinstalled with brew and it solved the problem.

KocaerBurak, 2024-06-19:
I have the same problem. My fan control doesnt work anymore