exelban/stats #1468: Question about how processes are measured/ weird issue about cpu usage of anecompiler

jpx40, 2023-05-13:
Hey, i wanna know how processes are messured how the cpu usages widget shows processes compared to Activity monitor.

Also to add i have weird issue. The anecompiler (Service for neural engine) is constanly running. Stats only shows that this processes is using my cpu (80 cpu usage)
, Activity monitor assumues that this process is in idle (0,2 cpu usage). I don't really know which monitor tool is showing the real usage. Maybe an malicious program using the service who knows. It shouldn't be this active at all.

jpx40, 2023-05-13:
Maybe it's fixed i enabled system integrity again. But still interested in this topic.

exelban, 2023-05-13:
information about processes is taken from the `top` utility.

closed 2023-05-15 by exelban