exelban/stats #1469: "Stats" cannot be opened because the developer cannot be verified

iasgharalikhan, 2023-05-15:
After performing a cask update using HomeBrew the app is not starting at restart.

**Details:**
 - Device: MacBook Pro 14 2021 M1 Pro
 - macOS: 13.3.1(a)
 - Application version: 2.8.21

iasgharalikhan, 2023-05-15:
[image]

dehydratedpotato, 2023-05-15:
@iasgharalikhan looks like something broke codesign during the update... As a solution until this is fixed, you can follow [these instructions from Apple](https://support.apple.com/lv-lv/guide/mac-help/mh40616/mac) to force launch it.

iasgharalikhan, 2023-05-15:
I tried it, there was no option to open anyway.
I had to move to bin and then reinstall it.

exelban, 2023-05-15:
Hi. I rechecked the last build. Everything is signed correctly.

exelban, 2023-05-15:
Could you post the output of `codesign -v -v /Volumes/Stats/Stats.app` (change the path if it's different)?

closed 2023-05-15 by exelban

iasgharalikhan, 2023-05-16:
Update: After the reinstallation everything works normally. the error is gone for now.

tonnylou44853, 2023-05-18:
I got the same problem after upgrading it to 2.8.21 via Homebrew (see picture below).

[image]

Here is the output of `codesign -v -v /Applications/Stats.app`:

```
/Applications/Stats.app: valid on disk
/Applications/Stats.app: satisfies its Designated Requirement
```