exelban/stats #1470: ssd % not reflecting correctly

crabsinabucket, 2023-05-16:
hey guys. absolutely love stats and recommend it to everybody, haven't had any issues up to now. usually ssd shows correctly the amount of space open on the hard drive, but suddenly it isn't. am i doing something incorrectly? i have already quit the process in activity monitor and restarted my machine, even did a fresh install of stats. 
[image]
amount stats says is available, seems to be stuck here
[image]
actual amount reflected in my system settings.

any idea what could be causing this?

exelban, 2023-05-16:
stats showing the disk usage with purgeable space. Because apple broke the API that allows it to get a free space without it.

closed 2023-05-16 by exelban

luckman212, 2023-09-22:
Is this still broken in Sonoma? Just wondering. I came here to report this "bug" but I see, aside from this, there is also:

- https://github.com/exelban/stats/issues/1589
- https://github.com/exelban/stats/issues/1552
- https://github.com/exelban/stats/issues/1510

So I obviously won't report again.

But, here's my screenshot anyway... from 13.5.2 + Stats 2.9.7

[image]

luckman212, 2023-09-22:
@exelban Is this API still broken in Sonoma? Just wondering. I came here to report this "bug" but I see, aside from this, there is also:

- https://github.com/exelban/stats/issues/1589
- https://github.com/exelban/stats/issues/1552
- https://github.com/exelban/stats/issues/1510

So I obviously won't report again.

But, here's my screenshot anyway... from 13.5.2 + Stats 2.9.7

[image]

exelban, 2023-09-22:
@luckman212 it's have nothing to issues that you mention. They are related to the "incorrect" free disk space size. I assume that you have selected this disk as primary in the settings?

luckman212, 2023-09-22:
@exelban Yes, that is my boot disk and it is selected as the main disk in settings. Should I move this to a new issue then?

Yesterday my Mac almost crashed because the disk filled up to 96% due to APFS snapshots going out of control. I think maybe because this space was technically "purgeable" maybe that's why the guage in Stats didn't report it?

I never had this bug before (nothing to do with Stats of course, just a macOS bug). I had to nuke all my local snapshots with 
```
tmutil deletelocalsnapshots /
```
Free space then went back to normal. Since then I also updated to 13.6.

exelban, 2023-09-22:
As I understand the space read is correct, but it does not match the widget value?
If yes, open a new issue for that. I will make a debug build for you.

luckman212, 2023-09-22:
Correct, space reading in the popup is correct, but widget has stopped updating or shows incorrect value. I will post a new issue, thank you