exelban/stats #1471: Question: Battery info section (wattage)

cakiya, 2023-05-16:
What does the battery wattage mean?
31 mA 
12.86 V
0.40 W <--is this 0.4w discharge or 0.4w charge?
31deg C <-- temperature of the battery
(macbook is plugged in)

Is my MacBook pulling 0.40w from the battery (discharging it), or is it charging at 0.40w?

 it would be really useful for seeing what workloads need how much watts to sustain (like, if i should bring my compact lower watt charger, or the really bulky brick that blocks other outlets)

[image]

tulpixer, 2023-05-21:
That's a good question. Let me add the question what's the difference between "Power: Battery" (3.63 W) and "Power: System Total" (2.01 W)

(MacBook not plugged in)
[image]

exelban, 2023-05-22:
@cakiya it's a power calculation based on voltage and current.
@tulpixer looks like the first one its battery power, and the other one it's the total power consumed by a system.

for more details, use google or ask apple for the meaning of these sensors. I have no idea what a lot of apple sensors mean.

closed 2023-05-22 by exelban