exelban/stats #1473: [Bug Report] Fan Manual Control

AShortcuts, 2023-05-19:
After every update of Stats, the manual control of my fan stops working. I am on thee latest version of Stats and macOS thus far and am using a MacBook Air 2018 (Intel). It fixes when I remove the fan helper and reinstall it. If you need more information, let me know. Thanks

f2sky, 2023-05-21:
I had the same problem with the Intel MacBook, every time the fan control function failed, the way to solve it was to restart the app.

rmilkowski, 2023-05-22:
Same issue on Mac Studio and Ventura 13.4 - adjusting fan speeds stopped working for sometime now (on recent betas and now on 13.4 too). Restarting app did not help. Uninstalling fan helper and then installing it again fixed the issue.

exelban, 2023-05-22:
duplicate of #1462

closed 2023-05-22 by exelban