exelban/stats #1489: Disk popup: SMART data doesn't always show

valynor, 2023-05-26:
Ventura 13.4, iMac 2017, v2.8.22

On my iMac the Disk popup sometimes looks like this: 

[image]

This is only _sometimes_ - if I quit and restart the app it usually shows the data correctly. 
I do not have this issue on my Mac Studio, on that machine it always displays the information.

exelban, 2023-05-26:
It means that SMART data is not available.

closed 2023-05-26 by exelban

valynor, 2023-05-26:
> It means that SMART data is not available.

I SPECIFICALLY wrote it only happens SOMETIMES. Most of the time it works and DOES SHOW values for "Temperature" and "Life"!

exelban, 2023-05-26:
`-` - means that data is not available. Maybe someone else at that time is reading SMART or something.