exelban/stats #1490: [Feature Request] Stop battery charging when upper limit is reached

hiroshi80, 2023-05-26:
## Suggestion
Charge to 80% to extend battery life. I would be very happy if it would automatically stop charging at 80%

The [Battery](https://github.com/actuallymentor/battery) seems to support

[image]

exelban, 2023-05-26:
I don't see any reason to duplicate the macOS internal feature.

closed 2023-05-26 by exelban

colinbrislawn, 2025-05-17:
Alternatives include AlDante (commercial) and [Battery-Toolkit](https://github.com/mhaeuser/Battery-Toolkit) (BSD-3, 1.2k stars) show community interest

I was curious if this package supports this, which is how I found this issue.