exelban/stats #1497: [Feature Request] Allow hiding certain modules from "combined modules"

heuristicAL, 2023-05-30:
It would be great if it was possible to hide modules from the menu bar but still have them shown in the popup. Right now, if I remove the widget from the module config, it doesn't show on the menu bar but it does take up the space that the widget would, if present.

Also: is there a way to show the full details of modules when the popup is open in the "combined module" mode? IE: see the full `CPU` module popup. That one might need to be another issue, I suspect it's a lot bigger lift 🙃

exelban, 2023-06-01:
Duplicate of #1189

exelban, 2023-06-02:
Nope, it cannot be done due to the specific of how the module work.
There are no plans to modify the combined modules popup view. Also, it's not possible to fit the full module popup when the combined view is enabled.

closed 2023-06-02 by exelban

doits, 2024-08-05:
I'm using this workaround: only display the label in menu bar.

For example I don't need sensors' info in the menu bar, but do want it in the popup. So I only display the label  `SEN` of the module. I takes a little bit of space in the menu bar but works otherwise.