exelban/stats #1512: Is there a way to reduce the space between the icons of the Stats app?

pmeaney, 2023-06-08:
After installing Stats, I see significant unused space between its icons.  Is there a way to reduce that space?
Thanks!

[image]

closed 2023-06-08 by exelban

moebis, 2025-01-20:
@pmeaney I discovered a cool glitch. If you turn on the combined module, and then turn it off, sometimes stats crashes (doesn't show the change), if you quit and then reload it, it will let you use the full size widgets, but with the combined spacing. Problem is I like the combined spacing, but I miss the details. You might have fiddle with it back and forth until you get it in a stuck state where it keeps the better spacing option. @exelban this should be included as an option.

[image]

moebis, 2025-01-20:
still shows 600mhz on M1 Ultra though. :-(