exelban/stats #1513: CPU widget not showing up at boot

d-graz, 2023-06-08:
**Describe the bug**
When booting up (after complete shutdown) the CPU widget is not showing up. Both RAM and NET widget are showing up as expected. Opening the Stats app and re-flipping the CPU toggle will solve the problem.
**Important Note** I'm running an Opencore Hackintosh but I strongly believe this should not represent a problem since the app just runs fine, in particular all the CPU sensors are displayed correctly.

**Details:**
 - Device: Macbook Pro 2019
 - macOS: 13.3.1
 - Application version: 2.8.24

Thank you for the support

exelban, 2023-06-10:
Hi. Thanks for the feedback.
Yeap, I have seen this behavior a few times. But cannot reproduce that. It's happening only after a boot up, and not always. Tried to catch that, but without success(

d-graz, 2023-06-10:
Yes, that is exactly what it’s happening (like on 3 boot out of 8). Let me know if there is any way to collect program logs so that I can share them here.

exelban, 2023-06-10:
Unfortunately, there are no logs from the app, only in the debug version that I share for some tests.
I am trying to catch that, but as I say without any success.

d-graz, 2023-06-10:
Unfortunately I cannot help you with code review since I'm not familiar with Swift and in general with MacOS. Maybe if you redirect me on some key classes of the project but as I said earlier I have no familiarity with Swift.

F1248, 2023-06-21:
I also have the problem that some widgets are not displayed after booting up. Restarting the app or reactivating the widget always helps.
For me the following widgets are affected:
- GPU (Bar Graph) sometimes, but not since macOS 13.4.1.
- 4 Sensors (Average / Hottest CPU / GPU) almost always
- Network (speed) sometimes 
- SSD, CPU, RAM and battery never

MacBook Air M1 (MacBookAir10,1)
macOS 13.4.1 (22F82)
Stats 2.8.26

exelban, 2023-06-26:
Hi.
I'm not sure what caused that problem. But right now I'm working on a totally new mechanism that turns on widgets. And I'm pretty sure that problem will disappear after.

closed 2023-06-26 by exelban