exelban/stats #1515: low battery notification doesn't work

alexander-potemkin, 2023-06-09:
[image]

It stopped working for some now...

closed 2023-06-10 by exelban

alexander-potemkin, 2023-06-10:
@exelban , is there something I missed? It used to work, but not anymore.
Are you removing this feature?

exelban, 2023-06-10:
for 99% you have disabled notifications from Stats. There were no changes for a long period of time.

alexander-potemkin, 2023-06-11:
Negative. There no changes on my side in there for a long time either...
[image]

exelban, 2023-06-11:
hmm, and it works before?

alexander-potemkin, 2023-06-11:
Yes, like a charm. System notification was appearing at 10%, I ignored it and I was waiting for iStats 5% notification.

But it didn’t appear. I was thinking that it might be me missed it, but when I started paying attention I realized it stopped from appearing…

My only change - it’s (stable) OS updates.

exelban, 2023-06-11:
first it's not iStats app.
Hmm, could you check if high level notification is working?

alexander-potemkin, 2023-06-12:
Apologies, I mixed names up!

High-level notification doesn't work either... To make sure it's not my attention issue, I've changed banners to alert, to make sure I have to close it explicitly.

alexander-potemkin, 2023-06-14:
I've got 100% charge notification today, but probably it's also a glue why other notifications didn't work - it says 13 mins, while the battery is fully charged actually.

Could it make some sense?

[image]

alexander-potemkin, 2023-06-22:
Triple checked - no low level notification; high level notification works.

alexander-potemkin, 2023-06-22:
@exelban , probably the ticket shall be re-opened?

exelban, 2023-06-23:
hi. I have checked that. And everything works fine. I receive a notification about the low battery level.

alexander-potemkin, 2023-06-23:
> hi. I have checked that. And everything works fine. I receive a notification about the low battery level.

Is there any debugging functionality available?

I understand you don't have this and can't reproduce - I wonder what I can do to nail it...

exelban, 2023-06-23:
There is no option to enable that. But I can say that if high notification works for you, low must work too. Because it's identical. 

You can try to send a low level notification, but you need to run the project and add logic for mocking the battery level.