exelban/stats #1518: Battery menu launch Top process as root

MrPhil17, 2023-06-13:
**Describe the bug**
After clicking on the battery icon to check the battery stats, Top process is launched as root, and it also keep change pid every second. After closing battery's menu, Top process quit.
This also drain a lot of battery because the CPU stay at around 15% for the whole time Top process is open.
This issue is present only on battery stats.
**Details:**
 - Device: MacBook Pro 15" mid 2014
 - macOS: 12.6.6
 - Application version: 2.8.25

exelban, 2023-06-13:
First of all, it's not possible `top` runs under the root. There is literally not presented in the app. And even more, it's not easy to run something as root.
Yes, `top` takes a lot of CPU when running. Nothing strange there.

MrPhil17, 2023-06-14:
Why top launches when I click on battery icon then?
I made a screen recording, this is the link:
https://github.com/exelban/stats/assets/72616599/654dcc35-8cc7-4096-95fd-0ed1ea059e2e

MrPhil17, 2023-06-14:
ok, but why CPU and network wont have the same behavior then? They both have processes lists, but wont launch top, as shown on the screen recording.

exelban, 2023-06-14:
because these modules do not use `top`