exelban/stats #1520: App quits often, once every day or two

jimbo1111, 2023-06-16:
The app keeps quitting for no particular reason, and I can't figure out why.

Couldn't find any crash reports in console, but I found this in diagnostic reports, seems to say something about a bug, and I see 3 reports in the last week, which meshes with about as often as the app seems to crash.

When it crashes, I don't notice any sort of dialog box or warning, it just disappears from the menu until I reopen it.  Sometimes upon reopening, the app doesn't show all modules, and I need to go in and toggle some modules off and on again, which is annoying.

{"app_name":"Stats","timestamp":"2023-06-16 14:57:27.00 -0700","app_version":"2.8.25","slice_uuid":"8a6ec78c-143c-332e-b47a-8d3d916e4f4d","build_version":"485","platform":1,"bundleID":"eu.exelban.Stats","share_with_app_devs":0,"is_first_party":0,"bug_type":"385","os_version":"macOS 13.3.1 (22E261)","roots_installed":0,"name":"Stats","incident_id":"6B02CB47-23BC-4DE0-9F95-3A592F7D97B1"}
{
  "uptime" : 1000000,
  "procRole" : "Default",
  "version" : 2,
  "userID" : 501,
  "deployVersion" : 210,
  "modelCode" : "MacBookPro18,4",
  "coalitionID" : 52952,
  "osVersion" : {
    "train" : "macOS 13.3.1",
    "build" : "22E261",
    "releaseType" : "User"
  },
  "captureTime" : "2023-06-16 14:57:24.1999 -0700",
  "incident" : "6B02CB47-23BC-4DE0-9F95-3A592F7D97B1",
  "pid" : 57704,
  "translated" : false,
  "cpuType" : "ARM-64",
  "roots_installed" : 0,
  "bug_type" : "385",
  "procLaunch" : "2023-06-13 16:36:11.7508 -0700",
  "procStartAbsTime" : 17939211276214,
  "procExitAbsTime" : 24017737190194,
  "procName" : "Stats",
  "procPath" : "\/Applications\/Stats.app\/Contents\/MacOS\/Stats",
  "bundleInfo" : {"CFBundleShortVersionString":"2.8.25","CFBundleVersion":"485","CFBundleIdentifier":"eu.exelban.Stats"},
  "storeInfo" : {"deviceIdentifierForVendor":"9BFEF7B2-C2E9-5F02-A1E3-288E3A7EF552","thirdParty":true},
  "parentProc" : "Exited process",
  "parentPid" : 57700,
  "coalitionName" : "eu.exelban.Stats",
  "crashReporterKey" : "D3AB4A43-780D-2E8C-24DC-C05DBC354AC4",
  "responsiblePid" : 57667,
  "codeSigningID" : "eu.exelban.Stats",
  "codeSigningTeamID" : "RP2S87B72W",
  "codeSigningFlags" : 570495745,
  "codeSigningValidationCategory" : 6,
  "codeSigningTrustLevel" : 0,
  "wakeTime" : 881678,
  "sleepWakeUUID" : "7B1A78BC-1251-4E4A-B6CD-4EBDC9AD2E1D",
  "sip" : "enabled",
  "exception" : {"codes":"0x000000000004aaaa, 0x0000000000000000","rawCodes":[305834,0],"type":"EXC_RESOURCE","signal":"SIGKILL"},
  "termination" : {"flags":2,"code":14123288431434181290,"namespace":"PORT_SPACE","indicator":"(Limit 305834 ports) Exceeded system-wide per-process Port Limit"},
  "extMods" : {"caller":{"thread_create":0,"thread_set_state":0,"task_for_pid":0},"system":{"thread_create":0,"thread_set_state":0,"task_for_pid":0},"targeted":{"thread_create":0,"thread_set_state":0,"task_for_pid":0},"warnings":0},
  "faultingThread" : 8,
  "threads" : [{"id":6931448,"queue":"com.apple.main-thread","frames":[{"imageOffset":3860,"symbol":"mach_msg2_trap","symbolLocation":8,"imageIndex":13},{"imageOffset":78400,"symbol":"mach_msg2_internal","symbolLocation":80,"imageIndex":13},{"imageOffset":39800,"symbol":"mach_msg_overwrite","symbolLocation":604,"imageIndex":13},{"imageOffset":4752,"symbol":"mach_msg","symbolLocation":24,"imageIndex":13},{"imageOffset":522424,"symbol":"__CFRunLoopServiceMachPort","symbolLocation":160,"imageIndex":14},{"imageOffset":516504,"symbol":"__CFRunLoopRun","symbolLocation":1208,"imageIndex":14},{"imageOffset":513420,"symbol":"CFRunLoopRunSpecific","symbolLocation":612,"imageIndex":14},{"imageOffset":200180,"symbol":"RunCurrentEventLoopInMode","symbolLocation":292,"imageIndex":15},{"imageOffset":199728,"symbol":"ReceiveNextEventCommon","symbolLocation":648,"imageIndex":15},{"imageOffset":199048,"symbol":"_BlockUntilNextEventMatchingListInModeWithFilter","symbolLocation":76,"imageIndex":15},{"imageOffset":233304,"symbol":"_DPSNextEvent","symbolLocation":636,"imageIndex":16},{"imageOffset":229620,"symbol":"-[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:]","symbolLocation":716,"imageIndex":16},{"imageOffset":181592,"symbol":"-[NSApplication run]","symbolLocation":464,"imageIndex":16},{"imageOffset":23496,"imageIndex":0},{"imageOffset":24360,"symbol":"start","symbolLocation":2236,"imageIndex":17}]},{"id":6931601,"name":"com.apple.CFSocket.private","frames":[{"imageOffset":49136,"symbol":"__select","symbolLocation":8,"imageIndex":13},{"imageOffset":692712,"symbol":"__CFSocketManager","symbolLocation":636,"imageIndex":14},{"imageOffset":28584,"symbol":"_pthread_start","symbolLocation":148,"imageIndex":18},{"imageOffset":7584,"symbol":"thread_start","symbolLocation":8,"imageIndex":18}]},{"id":6931624,"name":"com.apple.NSEventThread","frames":[{"imageOffset":3860,"symbol":"mach_msg2_trap","symbolLocation":8,"imageIndex":13},{"imageOffset":78400,"symbol":"mach_msg2_internal","symbolLocation":80,"imageIndex":13},{"imageOffset":39800,"symbol":"mach_msg_overwrite","symbolLocation":604,"imageIndex":13},{"imageOffset":4752,"symbol":"mach_msg","symbolLocation":24,"imageIndex":13},{"imageOffset":522424,"symbol":"__CFRunLoopServiceMachPort","symbolLocation":160,"imageIndex":14},{"imageOffset":516504,"symbol":"__CFRunLoopRun","symbolLocation":1208,"imageIndex":14},{"imageOffset":513420,"symbol":"CFRunLoopRunSpecific","symbolLocation":612,"imageIndex":14},{"imageOffset":1455368,"symbol":"_NSEventThread","symbolLocation":172,"imageIndex":16},{"imageOffset":28584,"symbol":"_pthread_start","symbolLocation":148,"imageIndex":18},{"imageOffset":7584,"symbol":"thread_start","symbolLocation":8,"imageIndex":18}]},{"id":9893441,"frames":[{"imageOffset":7564,"symbol":"start_wqthread","symbolLocation":0,"imageIndex":18}]},{"id":9904958,"frames":[{"imageOffset":7564,"symbol":"start_wqthread","symbolLocation":0,"imageIndex":18}]},{"id":9906431,"frames":[{"imageOffset":7564,"symbol":"start_wqthread","symbolLocation":0,"imageIndex":18}]},{"id":9906935,"frames":[{"imageOffset":7564,"symbol":"start_wqthread","symbolLocation":0,"imageIndex":18}]},{"id":9912368,"frames":[{"imageOffset":7564,"symbol":"start_wqthread","symbolLocation":0,"imageIndex":18}]},{"triggered":true,"id":9912369,"threadState":{"x":[{"value":0},{"value":57704},{"value":6132178476},{"value":105553161742832},{"value":105553161742848},{"value":9360},{"value":8704},{"value":5},{"value":8298022800,"symbolLocation":0,"symbol":"OBJC_CLASS_$___NSCFData"},{"value":8297955528,"symbolLocation":0,"symbol":"_current_pid"},{"value":36028797018963960},{"value":2043},{"value":2294466578},{"value":2045},{"value":2296565835},{"value":2294466578},{"value":18446744073709551572},{"value":8376708576},{"value":0},{"value":8297957216,"symbolLocation":0,"symbol":"mach_task_self_"},{"value":0},{"value":4401120992},{"value":0},{"value":4401121016},{"value":515},{"value":0},{"value":0},{"value":105553165288832},{"value":164}],"flavor":"ARM_THREAD_STATE64","lr":{"value":6818542728},"cpsr":{"value":536875008},"fp":{"value":6132179648},"sp":{"value":6132178416},"esr":{"value":1442840704,"description":" Address size fault"},"pc":{"value":6766329584,"matchesCrashFrame":1},"far":{"value":4400758784}},"queue":"eu.exelban.Stats","frames":[{"imageOffset":3824,"symbol":"task_name_for_pid","symbolLocation":8,"imageIndex":13},{"imageOffset":33928,"symbol":"__IOHIDEventSystemClientRefresh","symbolLocation":616,"imageIndex":19},{"imageOffset":32696,"symbol":"IOHIDEventSystemClientCreateWithType","symbolLocation":880,"imageIndex":19},{"imageOffset":7496,"symbol":"AppleSiliconSensors","symbolLocation":204,"imageIndex":3},{"imageOffset":55696,"imageIndex":3},{"imageOffset":445140,"imageIndex":6},{"imageOffset":413204,"imageIndex":6},{"imageOffset":59816,"imageIndex":6},{"imageOffset":17408,"symbol":"_dispatch_client_callout","symbolLocation":20,"imageIndex":20},{"imageOffset":30852,"symbol":"_dispatch_continuation_pop","symbolLocation":504,"imageIndex":20},{"imageOffset":110204,"symbol":"_dispatch_source_invoke","symbolLocation":1588,"imageIndex":20},{"imageOffset":47456,"symbol":"_dispatch_lane_serial_drain","symbolLocation":372,"imageIndex":20},{"imageOffset":50680,"symbol":"_dispatch_lane_invoke","symbolLocation":384,"imageIndex":20},{"imageOffset":94788,"symbol":"_dispatch_workloop_worker_thread","symbolLocation":648,"imageIndex":20},{"imageOffset":12404,"symbol":"_pthread_wqthread","symbolLocation":288,"imageIndex":18},{"imageOffset":7572,"symbol":"start_wqthread","symbo [... cut ...]
      "deploymentId" : 240000055
    },
    {
      "rolloutId" : "6297d96be2c9387df974efa4",
      "factorPackIds" : {

      },
      "deploymentId" : 240000008
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "c28e4ee6-1b08-4f90-8e05-2809e78310a3",
      "experimentId" : "6317d2003d24842ff850182a",
      "deploymentId" : 400000005
    },
    {
      "treatmentId" : "6dd670af-0633-45e4-ae5f-122ae4df02be",
      "experimentId" : "64406ba83deb637ac8a04419",
      "deploymentId" : 900000005
    }
  ]
}
}

exelban, 2023-06-17:
Have no idea what is that. Please share the crash report.

jimbo1111, 2023-06-17:
It doesn't generate a crash report as far as I can find

exelban, 2023-06-17:
only crash report could help

exelban, 2023-06-17:
if it crashes there must be a report. Otherwise, it's not a crash.

jimbo1111, 2023-06-17:
It's quitting on its own somehow, and I don't know how. I don't tell it to quit, it just disappears from the menubar. Console crash reports doesnt show anything from stats. If there is another place it stores crash reports I can try to find it.

exelban, 2023-06-18:
The console has a tab: `Crash reports` it must be there. There is no other place where it could be found. If there is no crash from Stats it mostly means that something just kills the Stats.

nedex, 2023-06-18:
Same problem here, grab these.

[Stats-2023-06-14-081747.ips.txt](https://github.com/exelban/stats/files/11781352/Stats-2023-06-14-081747.ips.txt)
[Stats-2023-06-14-233951.ips.txt](https://github.com/exelban/stats/files/11781353/Stats-2023-06-14-233951.ips.txt)
[Stats-2023-06-16-092712.ips.txt](https://github.com/exelban/stats/files/11781354/Stats-2023-06-16-092712.ips.txt)
[Stats-2023-06-16-200110.ips.txt](https://github.com/exelban/stats/files/11781355/Stats-2023-06-16-200110.ips.txt)
[Stats-2023-06-13-124419.ips.txt](https://github.com/exelban/stats/files/11781356/Stats-2023-06-13-124419.ips.txt)
[Stats-2023-06-12-104343.ips.txt](https://github.com/exelban/stats/files/11781357/Stats-2023-06-12-104343.ips.txt)

jimbo1111, 2023-06-18:
I am aware of the console tab crash reports, there is nothing in it from stats.

I don't know what else is or could be killing it, because I'm not killing it, it just disappears out of the menu bar.  This started relatively recently, maybe a month or two ago.  And the app closes, rather than being hidden on the bar (i could have thought it was Bartender hiding it or something, but nope, because reopening the app makes it show up again).

And separately there is the issue of some modules not showing when I restart the app, and me having to toggle them off and on again. Have not been able to find a pattern for this one.

jimbo1111, 2023-06-19:
It just quit again, no crash report, but it did make a similar diagnostic report as the one above. And the last 3 diagnostic reports have happened in exactly the same amount of time, which is odd, since the app has been closed for some portion of that time. But each is 2 days, 22 hours, 58 minutes apart (though doesn't seem to go down to the seconds)

[image]

nedex, 2023-06-20:
Any news? It keeps crashing

[Stats-2023-06-20-132048.ips.txt](https://github.com/exelban/stats/files/11800943/Stats-2023-06-20-132048.ips.txt)

exelban, 2023-06-20:
try with the disabled Network module.

jimbo1111, 2023-06-20:
Alright, I turned it off (was using them for network and disk), be back in 2 days and 4 hours to see if its broken heh

I do still have history turned on for cpu usage, but it's the single graph module rather than the double graph one

exelban, 2023-06-20:
Crash reports that are here have links to this specific widget. If that causes the crash we could dig in that way to find a problem. Otherwise I cannot help(

WayOfCon, 2023-06-20:
Also been having the same issue.

Grabbed these from the crash reports tab from console today, hopefully they can be of some use:

[Stats-2023-06-20-133750.ips.txt](https://github.com/exelban/stats/files/11807316/Stats-2023-06-20-133750.ips.txt)
[Stats-2023-06-20-133827.ips.txt](https://github.com/exelban/stats/files/11807317/Stats-2023-06-20-133827.ips.txt)
[Stats-2023-06-20-134007.ips.txt](https://github.com/exelban/stats/files/11807318/Stats-2023-06-20-134007.ips.txt)
[Stats-2023-06-20-133743.ips.txt](https://github.com/exelban/stats/files/11807319/Stats-2023-06-20-133743.ips.txt)

WayOfCon, 2023-06-20:
Seems to crash whenever I open the battery tab

jimbo1111, 2023-06-22:
It quit again, this time I was afk, and I had turned off those modules as mentioned.

It quit at 3:48pm so not the exact same amount of time as the last two. But perhaps it's the same amount of time since I changed the options, since I did that a couple hours after it quit last time around?  Or since I updated the app last time, since an update showed up recently?

Anyway, 3 days and 1hr 53mins after the last quit (again, no crash report) is when it quit this time. Seems to be pretty consistent around 3 days.

exelban, 2023-06-23:
ok, please share the crash report then.

jimbo1111, 2023-06-23:
It doesn't generate a crash report, just the same"diagnostic report" I posted above. Here's the most recent one

[ExcResource_Stats-2023-06-22-154857.txt](https://github.com/exelban/stats/files/11853467/ExcResource_Stats-2023-06-22-154857.txt)

exelban, 2023-06-24:
diagnostic report is useless.
I'm pretty sure there are a few different problems. And each member of this issue has a different reason for the crash. 

@jimbo1111 in your case I cannot help at all. Without a crash report, I don't have any information about what happened. Or even if Stats crash by themself since there is no crash report.

@nedex if you are interested in fixing your crash please open a seperate issue.

closed 2023-06-24 by exelban

jimbo1111, 2023-06-25:
Interesting - it just quit a lot earlier. When I reopened it, it was ready to update. I'm curious if the autoupdate checker has something to do with it?  Does it check every 3 days perhaps, and/or when an update comes out it gets pushed?  Because I think I recall this happening once before, coincident with an update.