exelban/stats #1522: Cannot get data of network download

huynq8, 2023-06-17:
Hi, 
I have just installed the newest version but it cannot get the value of network download.
Version 2.8.24

exelban, 2023-06-17:
Hi.
what do you mean?

huynq8, 2023-06-17:
Hi,
I'm using MAC OS Sonoma 14, and in the Network > download it does not
display the value. As the image below:
[image: Ảnh màn hình 2023-06-17 lúc 21.02.19.png]
Pls help me fix this bug. Thank you.

exelban, 2023-06-17:
there is no support for macos sonoma

closed 2023-06-17 by exelban

huynq8, 2023-06-18:
OK, thanks.
But I see you mentioned it supports "Stats is currently supported on macOS
10.14 (Mojave) and *higher*.". I hope will have stats for MAC OS 14 in the
future.

exelban, 2023-06-18:
I checked Sonoma and everything works fine.

yhx5768, 2023-09-20:
https://github.com/exelban/stats/issues/1579#issuecomment-1657917103
> switch to the process based reader

this works great for me