exelban/stats #1524: "Low power mode" doesn't update when switching via shortcut

peppy, 2023-06-18:
**Describe the bug**

I use a shortcut to switch lower power mode on and off via a hotkey:

https://www.icloud.com/shortcuts/e5ac3549b0074bd48ef4a6693abf48a9

When using this, the low power status and icon colour doesn't update. I'm pretty sure it used to a while back on a previous version of macOS or Stats.

Note that after a restart of Stats it reflects correctly.

https://github.com/exelban/stats/assets/191335/eb1b813d-ede9-4a39-aa5e-8b7e2b04349a

**Details:**
 - Device: MacBook Air M2 2022
 - macOS: Ventura 13.4
 - Application version: 2.8.25

closed 2023-06-18 by exelban

peppy, 2023-06-18:
Hey, thanks for the response.

Are you able to give a bit more detail here as to the resolution?

- Is this out of scope because I'm using a shortcut? If so, note that this bug also occurs when changing the low power mode setting from the apple settings panel (it doesn't update in Stats)
- If this is low priority and dev time cannot be assigned to it, but it can be agreed that it should be fixed, I will consider PRing a fix. Is that a valid direction?

exelban, 2023-06-18:
it's a broken Apple API that does not notify when the low mode is changed.

peppy, 2023-06-18:
Would polling be out of the question?

```sh
pmset -g | grep "lowpowermode" | tail -c 2
```

will return a `0`/`1` state for lower power mode (i've been using this in my script over multiple macOS releases.

[image]

peppy, 2023-06-18:
As you say, the API looks completely broken. Not only the notification, but the actual internal state doesn't update, even when plugging/unplugging the device.

I've made a patch which at least fixes the state not updating on next `read()`. In local testing this improves things significantly – while the state doesn't update immediately it will eventually, unlike before. I'm not immediately sure how often `read()` is triggered here as I'm not too familiar with the code base yet.

```diff
diff --git a/Modules/Battery/readers.swift b/Modules/Battery/readers.swift
index 19eacf0f..1f280a73 100644
--- a/Modules/Battery/readers.swift
+++ b/Modules/Battery/readers.swift
@@ -59,6 +59,39 @@ internal class UsageReader: Reader<Battery_Usage> {
     }
     
     public override func read() {
+        // Apple's low power mode state is completely broken.
+        // The NSProcessInfoPowerStateDidChange callback doesn't fire, and the information in
+        // IOPSCopyPowerSourcesInfo is not up-to-date.
+        //
+        // Reading directly from pmset is accurate though, so let's prefer that for now.
+
+        let task = Process()
+        task.launchPath = "/usr/bin/pmset"
+        task.arguments = ["-g"]
+
+        let outputPipe = Pipe()
+        defer {
+            outputPipe.fileHandleForReading.closeFile()
+        }
+        task.standardOutput = outputPipe
+
+        do {
+            try task.run()
+        } catch let err {
+        }
+
+        let outputData = outputPipe.fileHandleForReading.readDataToEndOfFile()
+        if let output = String(data: outputData, encoding: .utf8)
+        {
+            let lines = output.components(separatedBy: .newlines)
+            for line in lines {
+                if line.contains("lowpowermode") {
+                    self.usage.isLowPowerMode = line.last == "1"
+                    break
+                }
+            }
+        }
+
         let psInfo = IOPSCopyPowerSourcesInfo().takeRetainedValue()
         let psList = IOPSCopyPowerSourcesList(psInfo).takeRetainedValue() as [CFTypeRef]
         
@@ -73,9 +106,6 @@ internal class UsageReader: Reader<Battery_Usage> {
                 self.usage.isCharged = list[kIOPSIsChargedKey] as? Bool ?? false
                 self.usage.isCharging = self.getBoolValue("IsCharging" as CFString) ?? false
                 self.usage.optimizedChargingEngaged = list["Optimized Battery Charging Engaged"] as? Int == 1
-                if #available(macOS 12.0, *) {
-                    self.usage.isLowPowerMode = ProcessInfo.processInfo.isLowPowerModeEnabled
-                }
                 self.usage.level = Double(list[kIOPSCurrentCapacityKey] as? Int ?? 0) / 100
                 
                 if let time = list[kIOPSTimeToEmptyKey] as? Int {

```