exelban/stats #1532: [Feature request] Web-hook/API for server monitoring

MistyMoose, 2023-06-24:
Hi!

Thank you for your hard work; love the app. Increasingly Mac Mini/Studio/Pro are being run headless, so it would be super helpful to have a webhook or some sort of an API for Stats so the server can be monitored.

Thank you.

exelban, 2023-06-24:
Hi. Thanks for the feedback.
What do you mean by server can be monitored?

MistyMoose, 2023-06-24:
Absolutely. For headless systems, currently the only way to see the stats dashboard (I believe) is to use a Remote Desktop into the machine. If stats exposes the data on a specific port, whether it's in the form of a HTML page or something like that, it would allow you to monitor how the server is doing (CPU/GPU utilization, read/write to disks, etc...) from a webpage or a client rather than having to Remote Desktop into the machine. 

Basically, something like this: https://nicolargo.github.io/glances/

Stats does a really nice job of gathering all the data, so running something like Glances is redundant and unnecessary if stats could just have an option to "broadcast" the data. 

Does that make sense?

exelban, 2023-06-24:
oh, you mean monitoring the mac throw the network?

exelban, 2023-06-24:
Unfortunately it's not possible. At least for now. It could be possible when Stats starts exposing in some way collected data. Only after someone could use them to scrap and expose in some way. I'm focusing on local monitoring. Not external. 

TL.DR. when Stats will expose the data for 3d party tools it will be available to create a "plugin" like that. But it's not possible now.

closed 2023-06-24 by exelban