exelban/stats #1537: issues with iphone specific apps

metrocoding, 2023-06-28:
After introducing the M series chipset, you could run some apps designed for iPhone & iPad on Mac, for example, proxy apps like Shadowrocket.
When I run Shadowrocket, all my traffic goes through it and Stats app network indicator shows 0 upload and download however it works great with native macOS proxy apps.

exelban, 2023-06-28:
Stats read the data from the system. It looks like this app in some way go throw or not report the data. The problem is not in the Stats.

closed 2023-06-28 by exelban

metrocoding, 2023-06-28:
Sorry for the bad feedback.
App (Shadowrocket) report data is available in Activity Monitor even in Stats top processes, but the overall indicator shows zero.
Please take a look at the picture for more information.
If you need more information from me, please don't hesitate to let me know.
[image]

exelban, 2023-06-28:
in this case just switch the reader type to the `processes`

metrocoding, 2023-06-28:
Excellent, it works fine.
Thanks for your help.