exelban/stats #1538: Crash when clicking CPU widget

valynor, 2023-06-28:
Mac Studio, shortly after boot: I clicked the CPU widget to open the popup, Stats immediately crashed: 

[stats.txt](https://github.com/exelban/stats/files/11894356/stats.txt)

exelban, 2023-06-30:
Sorry, but this crash report does not help. Do you change some settings in the app before?
Or do you download the app and it crashes from the start?

valynor, 2023-06-30:
No, I've been using Stats for more than a year and it only crashed 3 or 4 times. 
All the crashes, like this one, happened very shortly (within 1 minute) after booting up macOS. So far it has *never* crashed later than that. I can only assume it has something to do with the OS not having finished booting completely or doing things in the background that cause Stats to be unstable.

exelban, 2023-06-30:
strange. I never faced a problem like that. Also, it cannot be reproduced. The only way to debug the problem it's experimentally, running the Stats each time enabling some feature. And find a feature that causes the crash. If you would like to fix that give me know, and I will take a look what can be checked first.

Also, did you use fan control?

valynor, 2023-07-01:
Since these crashes only happen once every 2-3 months I think it would really take too long to run it with some modules disabled, sorry. 

Yes, I use fan control but I only enabled this a few weeks ago. I had ~3 crashes without fan control too - this was the first one with fan control enabled.

exelban, 2023-07-01:
Yeap, it will take a lot of time. Still thanks for the feedback.

closed 2023-07-01 by exelban

valynor, 2023-07-08:
Got another crash directly after boot, stats showed up in menu bar and crashed 1 second later. 
Not posting the whole crashlog since it probably won't help. 

Time Awake Since Boot: 30 seconds

System Integrity Protection: enabled

Crashed Thread:        3  Dispatch queue: com.apple.root.background-qos

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0xfffffffffffffff8
Exception Codes:       0x0000000000000001, 0xfffffffffffffff8

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [726]

exelban, 2023-07-08:
Do you have the latest version?
I make a change in the thread QoS. Maybe it will help.

valynor, 2023-07-08:
Yes. 2.9.1