exelban/stats #1541: Help with UI/UX

exelban, 2023-06-30:
Hi.
I am looking for someone who will help with app design. The current is quite boring already.
Ready to hear any propositions.

Jameszjwchang, 2023-07-01:
Hello!
You could consider changing some of the views to swiftUI since they are a lot easier to maintain update in the future, and most of the vanilla UI already fits natively with the design of macOS. The design and prototyping process is also a lot more straightforward when coding. I could also slightly help with designing some of the views if you like.

Junyi-99, 2023-08-14:
As a long-time user of `stats`, I've been closely following its development. I've noticed that the functionality tends to change with each version update.

You mentioned that the UI is "quite boring." I assume you are referring to the following interface:
[image]

**To assist the stats' development, I believe addressing these questions will help us align our goals and ensure that the future updates of stats meet the needs and expectations of its users:**
1. Which parts of the UI/UX would you like to redesign? The `settings window`, `popup window`, 'menu bar'?
2. What system resources do you want **to emphasize displaying** in the new version of the UI? Should it primarily show numbers or include charts as well? Additionally, do you feel that all the current features are necessary, or are there any that could be reconsidered?
3. How do you want users to use your product? Is it intended for users to quickly glance at the general state of the system, or should it provide more detailed statistical information?

Thank you for your attention, and I look forward to your response.

Junyi-99, 2023-08-14:
Maybe we could develop a "`simple mode`" and an "`expert mode`". Users can choose between simplified or comprehensive modes to suit their needs.

For example, I think I'm a `simple mode` user, because I don't usually look at every metric in the "sensors panel", but rather focus on **the hottest components** (e.g. the hottest CPU), and the **input/output power** of my computer, and I usually focus on **utilization** instead of looking at how much resources are being used by specific processes. For "CPU and Memory" I usually focus on utilization and don't really look at how much resources are being used by specific processes.

On the other hand, the "`expert mode`" can retain the current UI design and provide the most accurate and detailed statistical data for users who require a deeper understanding of their system's performance.

I believe this approach would enhance the user experience and make the app more versatile.

GiorgioMendieta, 2023-09-06:
For the network pop-up widget, on the top processes I suggest replacing '0 KB/s' with '-' to reduce visual clutter, and maybe add an icon for upload/download (top and bottom arrow icons)

F1248, 2023-09-07:
> For the network pop-up widget, on the top processes I suggest replacing '0 KB/s' with '-' to reduce visual clutter, and maybe add an icon for upload/download (top and bottom arrow icons)

Arrow symbols can already be displayed under Network → Widgets → Pictogram → Arrows if the widgets "Speed" is activated.

sumitsahoo, 2023-11-06:
You can take reference from Sensei app, visually it is a stunner. 

[image]

[image]

vid, 2024-02-18:
Could I suggest making it informative? Some elements of stats are quite informative and it's possible to act on them, but it's not clear what they mean or what to do next. Memory is a particular area of confusion for users, including myself; I have no idea what "F U" means in one of the memory widgets, except it seems to be angry at me for some reason.
For simplicity's sake, maybe just link to a github pages page on that topic? Though enabling comments would be nice.

[image]

Thanks for your consideration!

Junyi-99, 2024-02-18:
> Could I suggest making it informative? Some elements of stats are quite informative and it's possible to act on them, but it's not clear what they mean or what to do next. Memory is a particular area of confusion for users, including myself; I have no idea what "F U" means in one of the memory widgets, except it seems to be angry at me for some reason. Pretty much every screen element could link to more information. For simplicity's sake, maybe just link to a github pages page on that topic? Though enabling comments would be nice.
> 
> <img alt="image" width="76" src="https://private-user-images.githubusercontent.com/68174/305718461-b26a9d04-f634-4dfd-973c-f7ea11b8d0bd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgyNjc1MDEsIm5iZiI6MTcwODI2NzIwMSwicGF0aCI6Ii82ODE3NC8zMDU3MTg0NjEtYjI2YTlkMDQtZjYzNC00ZGZkLTk3M2MtZjdlYTExYjhkMGJkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE4VDE0NDAwMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZlMWZkNDk3ZDY0ODIxYmM5NzMxYThmOWI2YThmMDFhM2ZjNmI4NDhkNTA3NDJhMmU3YjVlNGQ1M2ZlNTg2ZDImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.W3SllfP04idad_07gbMSv2Bo0rigeY9v8Otqk7AUNxk">
> Thanks for your consideration!

You have so much memory! 🤣

closed 2024-05-05 by exelban