exelban/stats #1544: [Feature Request] Show the type of interface used for Networkconnection

KlaraJa, 2023-07-02:
Hi,

I think it might be a really cool feature if you can see if your Mac is currently connected via ethernet, WiFi, USB or bluetooth.

exelban, 2023-07-03:
Hi. Yeap, you can do it in the network popup view.

closed 2023-07-03 by exelban

KlaraJa, 2023-07-12:
I meant as a new icon for the top menu.

Also with Wi-Fi connected it is only telling me unknown on my MacBook Pro M1 Pro (2021), is that a bug?

moritzlerch, 2023-08-02:
@exelban I would also be interested in this feature. I would be very nice if there would be an extra widget that is showing an icon for the connection type / network interface the Mac is currently connected to.

Like the original but e.g. with an ethernet icon. 

[image]

Ed1123, 2023-10-11:
I think this was incorrectly closed and should be reopened.