exelban/stats #1559: Auto-update doesn't check OS compatibility

rvwagner, 2023-07-14:
The auto-update system doesn't verify that the new version is compatible with the current OS before updating and relaunching.  With the default(?) "silent" update check mechanism, this means that on macOS 10.14.6 (wish I could update, but I'm locked to this version for some company IT reason), within a couple seconds of launching version 2.8.26, it updates, quits, and can't relaunch because the new version requires macOS 10.15.  I managed to work around this by launching from the read-only disk image in order to change the preferences to "never check for updates".

Obviously this can't be fixed for this round of dropping OS support, but if you could add in a compatibility check, that might save someone some trouble the next time you drop support for an ancient OS.

exelban, 2023-07-14:
Yes, there is a mechanism that pin releases to the specific macOS version. v2.8.26 - it's the last release that supports macOS 10.14. I have checked that right now and it works as expected.

closed 2023-07-14 by exelban

rvwagner, 2023-07-14:
Well, I'll just say that empirically, a copy of v2.8.26 running on 10.14.6 offers to update to v2.9.2 if I click "check for update", and if I set "check for updates" to "silent", it automatically upgrades to a newer version that isn't able to launch, so if that mechanism was supposed to be working in v2.8.26, it's broken on my machine.

exelban, 2023-07-16:
the only way is if your app receives the update through github,, not my server. Because in this case, I have no control over what version will be returned. The latest release always will be returned.

rvwagner, 2023-07-18:
That seems likely.  Digging into the code a bit, it should definitely be contacting your server, but I think it's running into an SSL failure and falling back on GitHub, probably because of an expired pinned certificate in the OS that you can't do anything about (I've had this issue with another app's updater/registration check).

```
$ curl https://api.serhiy.io/v1/stats/release/latest
curl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to api.serhiy.io:443
```

Running the same command on a newer OS, I get "connection reset by peer" instead, which seems much more appropriate for that likely-malformed request.

exelban, 2023-07-18:
there is nothing to server. Most probably your network blocking server or something wrong with SSL on your machine. Because tls for this server is ensured by Cloudflare. 

Yeap, if the app cannot fetch the last release from the server it will go to the github as a fallback.

exelban, 2023-07-18:
thanks. Solution for you it's to disable the auto-update.
If you cannot do it because there is not enough time before the app is updated and stops working you can reset the settings with cli: `defaults delete eu.exelban.Stats`

It will reset all settings for the Stats and you can reinitialize the Stats from scratch with disabled auto-updates.

mrkapqa, 2024-05-03:
you would need to termporariy disable Internet though, because atleast one my computer, once the Download is initiated (on Mojave) , it would re-install/reapply it automatically, even with after defaults delete eu.exelban.Stats initiated, it somehow fetches it from cache, and tries to install it;

disabling Internet gives you time to set the Auto-update properly to OFF.

rvwagner, 2024-05-03:
That matches my experience.  The default setting for Stats is apparently to automatically and silently update, so deleting the preferences doesn't help.

If you *can't* disable your internet for some reason (as in the case of the company computer with network user accounts I was working on), a workaround is to launch Stats from the disk image you downloaded it on (telling MacOS/Stats *not* to move it to the Applications folder, if it offers).  Because that's a read-only volume, Stats doesn't do the automatic update-and-relaunch process, so you have all the time you need to change the settings.

This works even if you already have a copy of Stats installed in your Applications folder, no need to delete anything first.