exelban/stats #1567: [Feature Request] Override mode to allow to turn off fans

alexandresoro, 2023-07-22:
Hi,

I'd like to suggest what seems to be a strange request by looking only at the title.

I currently own a Mac Studio that by default allows to set up fan speed between 1100 RPM and 3500 RPM. 
This works fine with the sensors popup.

However, I'd like to suggest the possibility to completely override the minimum speed and basically turn the fans off (0 RPM).
So the exact opposite of the current "turbo mode".

### Context of this request 

My Mac is affected by a documented high-pitched noise (see for example https://www.macrumors.com/2022/04/28/mac-studio-high-pitched-noise/ ) which can be really annoying.
After investigation, this is due to exhaust holes "singing" to the tune of the fans.
Mac Studio fans are always on and around 1300RPMs regardless of the load or idling.

So there's definitely room to turn them off safely.

Technically speaking, it seems feasible as I currently use a script to achieve this: https://github.com/dirkschreib/SilentStudio .
I've been using this script for months without any issue, and I believe that this feature could fit with the existing fans control feature.

I understand that this might harm some setup if used improperly, so maybe this would need some kind of disclaimer like "use this at your own risk."

exelban, 2023-07-27:
Hi.
Hmm, I have no this problem on the mac studio, fortunately)
But I see the problem and I can add an option to stop the fan for start. For now, I have no plans to add curves for fan control. Maybe in the future I will add this option.

closed 2023-08-06 by exelban

valynor, 2023-08-06:
Hmm, I think this is a nice feature but with some potential for disaster, especially on older, hotter Intel Macs. 
I'd like to suggest that you add a confirmation dialog "Do you really want to turn off the fans? Your device might overheat/get damaged" (or something like this) to the fan-off button.

Junyi-99, 2023-08-15:
Appreciate the post, after reading your post my MacBook Pro 2019's fan had a strange rattling noise, this script helped me to turn off the fan so I could have a moment of peace 😂

valynor, 2023-08-15:
Typo: `alert.informativeText = localizedString("You are going to turn off the fan. That is not recommended action that can broke your mac, are you sure you want to do that?")`

broke -> break