exelban/stats #1568: Add feature to show ECORE/PCORE CPU colors running on MACOS Intel Hackintosh using Alder Lake/Raptor lake CPU

rogeriomm, 2023-07-23:
"It would be great if the Popup window on a MACOS Intel Hackintosh running Alder Lake or Raptor Lake CPU could display the ECORE/PCORE CPU colors."

exelban, 2023-07-27:
it's not related to the Stats. In this case, you need to have efficient/performance cores identification in your SMC. Stats will automatically select cores If macos will see the efficient/performance cores.

closed 2023-07-27 by exelban

rogeriomm, 2023-07-27:
> it's not related to the Stats. In this case, you need to have efficient/performance cores identification in your SMC. Stats will automatically select cores If macos will see the efficient/performance cores.

Unfortunately Apple doesn't support assymetric CPU's on the MAC OS Intel, Hackintosh is an unauthorized hacking. I think that the hacincking must try mimic the original Intel Apple, at least in kernel level. At application level on Stats it'd be ok

exelban, 2023-07-28:
in Hackintosh you have a custom SMC (the last I use was FakeSMC). And it provides all information about the hardware to the macos. Pretty sure it's possible to add as there is separation for efficient/performance core in the io.