exelban/stats #1578: Persistent crashing on wake after OS sleep

nfaux, 2023-07-30:
Very time I wake, after the system sleeps, i receive the below crash report.
Stat Version 2.9.2
MacBook Pro 16inch - OSX 13.5 (22G74)

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [7624]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.9.2 (491)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               502

Date/Time:             2023-07-30 13:50:13.9194 +1000
OS Version:            macOS 13.5 (22G74)
Report Version:        12
Bridge OS Version:     7.6 (20P6072)
Anonymous UUID:        DE360C6D-A8DB-12F4-7859-D37410EB77AD

Sleep/Wake UUID:       CADF14AC-54D4-40BA-92BD-A14F2595AE6B

Time Awake Since Boot: 200000 seconds
Time Since Wake:       193091 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 4 Illegal instruction: 4
Terminating Process:   exc handler [7624]

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   Kit                           	       0x10f8947df 0x10f885000 + 63455
1   Kit                           	       0x10f895128 0x10f885000 + 65832
2   AppKit                        	    0x7ff8046b2690 _NSViewDrawRect + 121
3   AppKit                        	    0x7ff804e922fa -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 1810
4   AppKit                        	    0x7ff8046b1dbf -[NSView(NSLayerKitGlue) _drawViewBackingLayer:inContext:drawingHandler:] + 753
5   QuartzCore                    	    0x7ff808def065 -[CALayer renderInContext:] + 1761
6   QuartzCore                    	    0x7ff808df148e -[CALayer _renderSublayersInContext:] + 499
7   QuartzCore                    	    0x7ff808def074 -[CALayer renderInContext:] + 1776
8   QuartzCore                    	    0x7ff808df148e -[CALayer _renderSublayersInContext:] + 499
9   QuartzCore                    	    0x7ff808def074 -[CALayer renderInContext:] + 1776
10  QuartzCore                    	    0x7ff808df148e -[CALayer _renderSublayersInContext:] + 499
11  QuartzCore                    	    0x7ff808def074 -[CALayer renderInContext:] + 1776
12  QuartzCore                    	    0x7ff808df148e -[CALayer _renderSublayersInContext:] + 499
13  QuartzCore                    	    0x7ff808def074 -[CALayer renderInContext:] + 1776
14  AppKit                        	    0x7ff804e915f0 -[NSView _renderLayerInContext:] + 285
15  AppKit                        	    0x7ff804e91ea8 -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 704
16  AppKit                        	    0x7ff804e8ac67 __47-[NSView displayRectIgnoringOpacity:inContext:]_block_invoke + 843
17  AppKit                        	    0x7ff8045aff47 NSPerformVisuallyAtomicChange + 132
18  AppKit                        	    0x7ff804819462 -[NSView displayRectIgnoringOpacity:inContext:] + 85
19  AppKit                        	    0x7ff8048193bc __62-[NSView cacheDisplayInRect:toBitmapImageRep:includeSubviews:]_block_invoke + 295
20  AppKit                        	    0x7ff804819281 -[NSBitmapImageRep _captureDrawing:] + 56
21  AppKit                        	    0x7ff80481923c -[NSView cacheDisplayInRect:toBitmapImageRep:includeSubviews:] + 154
22  AppKit                        	    0x7ff804d2782e -[NSStatusItem _updateReplicant:] + 1384
23  AppKit                        	    0x7ff8047e243d -[NSStatusItem _updateReplicants] + 243
24  AppKit                        	    0x7ff804d283b6 -[NSStatusItem _updateReplicantsUnlessMenuIsTracking:] + 121
25  Foundation                    	    0x7ff802374d85 __NSFireDelayedPerform + 440
26  CoreFoundation                	    0x7ff8014e3358 __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 20
27  CoreFoundation                	    0x7ff8014e2ed3 __CFRunLoopDoTimer + 807
28  CoreFoundation                	    0x7ff8014e2af9 __CFRunLoopDoTimers + 285
29  CoreFoundation                	    0x7ff8014c8e39 __CFRunLoopRun + 2206
30  CoreFoundation                	    0x7ff8014c7f31 CFRunLoopRunSpecific + 560
31  HIToolbox                     	    0x7ff80af48f3d RunCurrentEventLoopInMode + 292
[... 609 lines cut ...]

Model: MacBookPro16,1, BootROM 1968.140.7.0.0 (iBridge: 20.16.6072.0.0,0), 8 processors, 8-Core Intel Core i9, 2.3 GHz, 16 GB, SMC 
Graphics: Intel UHD Graphics 630, Intel UHD Graphics 630, Built-In
Graphics: AMD Radeon Pro 5500M, AMD Radeon Pro 5500M, PCIe, 4 GB
Display: Color LCD, 3072 x 1920 Retina, MirrorOff, Online
Display: HP LA2306, 1080 x 1920, MirrorOff, Online
Display: C32JG5x, 2560 x 1440 (QHD/WQHD - Wide Quad High Definition), Main, MirrorOff, Online
Memory Module: BANK 0/ChannelA-DIMM0, 8 GB, DDR4, 2667 MHz, SK Hynix, HMA81GS6CJR8N-VK
Memory Module: BANK 2/ChannelB-DIMM0, 8 GB, DDR4, 2667 MHz, SK Hynix, HMA81GS6CJR8N-VK
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x7BF), wl0: Dec  9 2022 17:02:25 version 9.30.492.0.32.5.87 FWID 01-e7856862
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB 2.0 BILLBOARD
USB Device: T2Bus
USB Device: composite_device
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5

exelban, 2023-08-01:
Hmm, I think I know the problem. Will try to fix it.

JulindM, 2023-08-04:
Can confirm. Here is a log from me as well below.

<details>
<summary> Log </summary>
Process:               Stats [616]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.9.3 (492)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-08-04 12:47:20.6558 +0200
OS Version:            macOS 13.5 (22G74)
Report Version:        12
Anonymous UUID:        73260A63-ED18-299E-B090-BE6E8D8EFFA5

Time Awake Since Boot: 6500 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00004cf1bf9d3640
Exception Codes:       0x0000000000000001, 0x00004cf1bf9d3640

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [616]

VM Region Info: 0x4cf1bf9d3640 is not in any region.  Bytes after previous region: 84120149243457  Bytes before following region: 20951930685888
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      commpage (reserved)        1000000000-7000000000   [384.0G] ---/--- SM=NUL  ...(unallocated)
--->  GAP OF 0x5f9000000000 BYTES
      MALLOC_NANO              600000000000-600008000000 [128.0M] rw-/rwx SM=PRV  

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libobjc.A.dylib               	       0x19f957f00 objc_retain + 16
1   libswiftCore.dylib            	       0x1aedeedd4 objcRetainAndReturn(void*) + 32
2   Kit                           	       0x102689508 0x1025e8000 + 660744
3   libswiftCore.dylib            	       0x1aed859b8 initializeWithCopy for ClosedRange<>.Index + 196
4   Kit                           	       0x102689194 0x1025e8000 + 659860
5   Kit                           	       0x1026834d0 0x1025e8000 + 636112
6   Kit                           	       0x1026835e0 0x1025e8000 + 636384
7   AppKit                        	       0x1a3111850 _NSViewDrawRect + 124
8   AppKit                        	       0x1a37c23a8 -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 1088
9   AppKit                        	       0x1a31110c4 -[NSView(NSLayerKitGlue) _drawViewBackingLayer:inContext:drawingHandler:] + 556
10  QuartzCore                    	       0x1a72d9db4 -[CALayer _display] + 832
11  AppKit                        	       0x1a3110dcc -[_NSBackingLayer display] + 352
12  AppKit                        	       0x1a30a2894 -[_NSViewBackingLayer display] + 532
13  QuartzCore                    	       0x1a72d9264 CA::Layer::display_if_needed(CA::Transaction*) + 760
14  QuartzCore                    	       0x1a744cbc4 CA::Context::commit_transaction(CA::Transaction*, double, double*) + 456
15  QuartzCore                    	       0x1a72bba70 CA::Transaction::commit() + 648
16  AppKit                        	       0x1a312089c __62+[CATransaction(NSCATransaction) NS_setFlushesWithDisplayLink]_block_invoke + 272
17  AppKit                        	       0x1a37fccfc ___NSRunLoopObserverCreateWithHandler_block_invoke + 64
18  CoreFoundation                	       0x19fdd19f0 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
19  CoreFoundation                	       0x19fdd18dc __CFRunLoopDoObservers + 532
20  CoreFoundation                	       0x19fdd0f14 __CFRunLoopRun + 776
21  CoreFoundation                	       0x19fdd04b8 CFRunLoopRunSpecific + 612
22  HIToolbox                     	       0x1a9622df0 RunCurrentEventLoopInMode + 292
23  HIToolbox                     	       0x1a9622c2c ReceiveNextEventCommon + 648
24  HIToolbox                     	       0x1a9622984 _BlockUntilNextEventMatchingListInModeWithFilter + 76
25  AppKit                        	       0x1a2ff797c _DPSNextEvent + 636
26  AppKit                        	       0x1a2ff6b18 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716
27  AppKit                        	       0x1a2feaf7c -[NSApplication run] + 464
28  Stats                         	       0x102158e8c 0x102154000 + 20108
29  dyld                          	       0x19f99bf28 start + 2236

Thread 1:: com.apple.CFSocket.private
0   libsystem_kernel.dylib        	       0x19fcbf030 __select + 8
[... 472 lines cut ...]
      "deploymentId" : 240000007
    },
    {
      "rolloutId" : "5ffde50ce2aacd000d47a95f",
      "factorPackIds" : {

      },
      "deploymentId" : 240000303
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "a092db1b-c401-44fa-9c54-518b7d69ca61",
      "experimentId" : "64a844035c85000c0f42398a",
      "deploymentId" : 400000016
    },
    {
      "treatmentId" : "3a3cf641-8471-4e4e-9ad4-81d0ede970fd",
      "experimentId" : "64a84dae90d82611a0bd7d3d",
      "deploymentId" : 400000010
    }
  ]
}
}
</details>

ptoyip, 2023-08-05:
same here,macOS Ventura (13.3.0), M2 chips

PoMkaNik, 2023-08-09:
Confirm. Adding to login items not helped

closed 2023-08-21 by exelban

IsaacNez, 2023-11-06:
I keep having this issue on v2.9.10, although it is under Login Items. I am using the Summarized Widget.

exelban, 2023-11-06:
@IsaacNez please share the crash report

IsaacNez, 2023-11-06:
Here it is:
<details><summary>Logs</summary>
<p>
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [867]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.9.10 (505)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-11-03 19:23:31.6911 +0100
OS Version:            macOS 14.1 (23B74)
Report Version:        12
Anonymous UUID:        65DD21D2-D665-2940-D81A-34BCA4093AE0

Sleep/Wake UUID:       7B6F2BB7-6F18-411B-92EE-55394DF0E853

Time Awake Since Boot: 8000 seconds
Time Since Wake:       2632 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x0000000192b6d784

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [867]

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libswiftCore.dylib            	       0x192b6d784 swift_unknownObjectRetain + 44
1   libswiftCore.dylib            	       0x19285a820 String._bridgeToObjectiveCImpl() + 96
2   Kit                           	       0x10320adb4 0x103174000 + 617908
3   Kit                           	       0x10320a36c 0x103174000 + 615276
4   Kit                           	       0x10320a5f8 0x103174000 + 615928
5   AppKit                        	       0x1868e7b94 _NSViewDrawRect + 124
6   AppKit                        	       0x18725a848 -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 1088
7   AppKit                        	       0x1868e7408 -[NSView(NSLayerKitGlue) _drawViewBackingLayer:inContext:drawingHandler:] + 556
8   AppKit                        	       0x186dffc44 -[NSViewBackingLayer drawInContext:] + 56
9   AppKit                        	       0x186fdda98 -[NSViewBackingStore update:] + 560
10  AppKit                        	       0x186dfee40 -[NSViewBackingLayerContents update:] + 52
11  AppKit                        	       0x186dff878 -[NSViewBackingLayer display] + 1240
12  QuartzCore                    	       0x18b0fb278 -[CALayer renderInContext:] + 140
13  QuartzCore                    	       0x18b0fcf0c -[CALayer _renderSublayersInContext:] + 340
14  QuartzCore                    	       0x18b0fb6a8 -[CALayer renderInContext:] + 1212
15  QuartzCore                    	       0x18b0fcf0c -[CALayer _renderSublayersInContext:] + 340
16  QuartzCore                    	       0x18b0fb6a8 -[CALayer renderInContext:] + 1212
17  QuartzCore                    	       0x18b0fcf0c -[CALayer _renderSublayersInContext:] + 340
18  QuartzCore                    	       0x18b0fb6a8 -[CALayer renderInContext:] + 1212
19  QuartzCore                    	       0x18b0fcf0c -[CALayer _renderSublayersInContext:] + 340
20  QuartzCore                    	       0x18b0fb6a8 -[CALayer renderInContext:] + 1212
21  AppKit                        	       0x187259f48 -[NSView _renderLayerInContext:] + 204
22  AppKit                        	       0x18725a614 -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 524
23  AppKit                        	       0x187254908 __47-[NSView displayRectIgnoringOpacity:inContext:]_block_invoke + 656
24  AppKit                        	       0x186810ab0 NSPerformVisuallyAtomicChange + 108
25  AppKit                        	       0x186a17748 -[NSView displayRectIgnoringOpacity:inContext:] + 104
26  AppKit                        	       0x186a17690 __62-[NSView cacheDisplayInRect:toBitmapImageRep:includeSubviews:]_block_invoke + 192
27  AppKit                        	       0x186a175ac -[NSBitmapImageRep _captureDrawing:] + 72
28  AppKit                        	       0x186a17548 -[NSView cacheDisplayInRect:toBitmapImageRep:includeSubviews:] + 168
29  AppKit                        	       0x186f9d7dc -[NSStatusItem _updateReplicant:] + 1168
30  AppKit                        	       0x1869e78e8 -[NSStatusItem _updateReplicants] + 172
31  AppKit                        	       0x186f9e17c -[NSStatusItem _updateReplicantsUnlessMenuIsTracking:] + 96
32  Foundation                    	       0x18411a850 __NSFireDelayedPerform + 372
33  CoreFoundation                	       0x183011cd0 __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 32
[... 507 lines cut ...]
      },
      "deploymentId" : 240000502
    },
    {
      "rolloutId" : "62745b8e85854550ad70b6e4",
      "factorPackIds" : {
        "SIRI_TTS_DEVICE_TRAINING" : "650a28170efd3c4addfc7355"
      },
      "deploymentId" : 240000073
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "a092db1b-c401-44fa-9c54-518b7d69ca61",
      "experimentId" : "64a844035c85000c0f42398a",
      "deploymentId" : 400000019
    }
  ]
}
}

</p>
</details> 

Please let me know if there is any other way I can help.

AAnzel, 2024-02-15:
Hi there! First, thanks for this wonderful tool. Second, I am still experiencing this issue after logging in. The program sometimes starts but more than often crashes and requires a manual start. I added the program to autologin items. Below is the crash log.
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [870]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.9.20 (520)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               502

Date/Time:             2024-02-15 09:23:33.1038 +0100
OS Version:            macOS 13.6.4 (22G513)
Report Version:        12
Anonymous UUID:        11843A58-4521-F36F-3887-B916F8D13EEB

Time Awake Since Boot: 150 seconds

System Integrity Protection: enabled

Crashed Thread:        7  Dispatch queue: com.apple.NSURLSession-delegate

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   Stats [870]

Application Specific Information:
abort() called

Application Specific Backtrace 0:
0   CoreFoundation                      0x0000000193b970e4 __exceptionPreprocess + 176
1   libobjc.A.dylib                     0x00000001936b5fd0 objc_exception_throw + 60
2   CoreFoundation                      0x0000000193bc0330 _CFBundleGetValueForInfoKey + 0
3   AppKit                              0x0000000196d674d8 -[NSWindow(NSWindow_Theme) _postWindowNeedsToResetDragMarginsUnlessPostingDisabled] + 320
4   AppKit                              0x0000000196d557ac -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 816
5   AppKit                              0x0000000196eafb88 -[NSPanel _initContent:styleMask:backing:defer:contentView:] + 48
6   AppKit                              0x0000000196d55470 -[NSWindow initWithContentRect:styleMask:backing:defer:] + 48
7   AppKit                              0x0000000196eafb3c -[NSPanel initWithContentRect:styleMask:backing:defer:] + 48
8   AppKit                              0x0000000196d541d0 -[NSWindowTemplate nibInstantiate] + 216
9   AppKit                              0x0000000196d270a8 -[NSIBObjectData instantiateObject:] + 212
10  AppKit                              0x0000000196d26a74 -[NSIBObjectData nibInstantiateWithOwner:options:topLevelObjects:] + 252
11  AppKit                              0x0000000196d1c830 loadNib + 340
12  AppKit                              0x0000000196d1be60 +[NSBundle(NSNibLoading) _loadNibFile:nameTable:options:withZone:ownerBundle:] + 560
13  AppKit                              0x0000000196d1bb64 -[NSBundle(NSNibLoading) loadNibNamed:owner:topLevelObjects:] + 180
14  AppKit                              0x0000000196fed744 -[NSAlert init] + 92
15  Kit                                 0x000000010289ec58 block_destroy_helper.97 + 47640
16  Kit                                 0x00000001028b33a4 block_destroy_helper + 8312
17  Stats                               0x0000000102417c94 Stats + 162964
18  Kit                                 0x00000001028b4858 __swift_memcpy33_8 + 1432
19  Kit                                 0x00000001028b2ae4 block_destroy_helper + 6072
20  Kit                                 0x00000001028b2e48 block_destroy_helper + 6940
21  CFNetwork                           0x000000019836fa88 CFURLCredentialStorageCopyAllCredentials + 18940
22  CFNetwork                           0x0000000198387b80 _CFHTTPMessageSetResponseProxyURL + 12332
23  libdispatch.dylib                   0x000000019388e874 _dispatch_call_block_and_release + 32
24  libdispatch.dylib                   0x0000000193890400 _dispatch_client_callout + 20
25  libdispatch.dylib                   0x0000000193897a88 _dispatch_lane_serial_drain + 668
26  libdispatch.dylib                   0x000000019389862c _dispatch_lane_invoke + 436
27  libdispatch.dylib                   0x00000001938a3244 _dispatch_workloop_worker_thread + 648
28  libsystem_pthread.dylib             0x0000000193a3c074 _pthread_wqthread + 288
29  libsystem_pthread.dylib             0x0000000193a3ad94 start_wqthread + 8

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x1939ffef4 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x193a12220 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x193a08b58 mach_msg_overwrite + 604
[... 486 lines cut ...]
  "logWritingSignature" : "e9c47a95985f4d878627386670fda5cd9f3c505b",
  "trialInfo" : {
  "rollouts" : [
    {
      "rolloutId" : "5fb4245a1bbfe8005e33a1e1",
      "factorPackIds" : {

      },
      "deploymentId" : 240000021
    },
    {
      "rolloutId" : "60da5e84ab0ca017dace9abf",
      "factorPackIds" : {

      },
      "deploymentId" : 240000008
    }
  ],
  "experiments" : [

  ]
}
}

```