exelban/stats #1584: Update doesn't install after download

Saratxe, 2023-08-08:
**Describe the bug**
Stats notify me about an update. It downloads up to 100% but didn't start the update process after that. I've to download manually the DMG with the latest version and install it.

**Details:**
 - iMac Retina 5K, 27-inch, 2017, 3,8 GHz Quad-Core Intel Core i5
 - macOS: Ventura 13.5
 - Application version: 2.9.3 and some versions earlier (2.9.1 2.9.2)

exelban, 2023-09-02:
Hi. Thanks for the feedback.
Please check if there is a Stats.dmg file in the Downloads folder when it stacks on 100%.

Saratxe, 2023-09-11:
Hi!
Today Stats want to update to V 2.9.7, Download Window says 100% but there's no Stats.img in the Download folder.

closed 2023-10-21 by exelban