exelban/stats #1585: Battery Health is shown Inconsistency

SangTruongTan, 2023-08-10:
Hi Team,
I'm now facing an issue with the battery's health. In the System Information center, my MacBook says the battery's healthy is perfectly 100%. However, there is a discrepancy, the app says only 98% and 97% currently. Please refer to the illustration and my Mac's information below.
Note that the cycle count is correct.
[image]
[image]

**Details:**
 - Device: MacBook Pro M1 16
 - macOS: MacOS Sonoma 14.0
 - Application version: 2.9.3

closed 2023-08-10 by exelban

Junyi-99, 2023-08-15:
The `MaxCapacity` of your battery is `8482mAh`, The `DesignCapacity` of your battery is `8694mAh`,
Hence, the Health of your battery is `8482/8694 = 0.9756 = 98%`

There is no wrong with the `stats`, Mac's Information may shows you some incorrect information.

Here's a screenshot of my computer's battery status, hopefully you can understand the calculations behind it
[image]
In my computer, the battery health is calculated by `7247`/`8790 = 0.8245 = 82%`

You can use `ioreg -l -w0 | grep Capacity` to retrieve your battery information

SangTruongTan, 2023-08-16:
Thanks for your explanation, @Junyi-99 
[image]
It looks very weird to me since the app is fairly showing the same thing with `ioreg -l -w0 | grep Capacity`. However, the MaxCapacity is `100`. Is there a possibility that battery miscalibration causes this problem? Because it seems the maxCapacity drains down too fast thesedaysy when I keep plugging my Mac.