exelban/stats #1593: How to refresh Preference in Stats when edit plist in Cli Terminal with defaults write

9840380, 2023-08-21:
```
killall -kill Stats
defaults write eu.exelban.Stats fan_0_mode -int 1
defaults write eu.exelban.Stats fan_0_speed -int 6156
defaults write eu.exelban.Stats fan_1_mode -int 1
defaults write eu.exelban.Stats fan_1_speed -int 5700
defaults write eu.exelban.Stats Sensors_fanControl -bool YES
killall -u $USER cfprefsd
open -a Stats
```

How to refresh Preference?

exelban, 2023-08-21:
as I know it's enough to restart the app, it will load new preferences.
I have no idea what is the option, I don't see it in the app.

closed 2023-08-21 by exelban

9840380, 2023-08-21:
Thanks for so quick answer,-but no it doesn't work at all!!! When you set setting from terminal it doesn't refresh it on the app.
Part 1 check
[image]

Part 2 Demonstrate
[image]

9840380, 2023-08-21:
> as I know it's enough to restart the app, it will load new preferences.
> I have no idea what is the option, I don't see it in the app.

Here is my investigation

And i don't find any option to set max speed fun from plist. It can be done only from GUI.
It can be useful when automation something, or new system...
[image]

exelban, 2023-08-21:
sorry, but I will not even try to understand what do you want to achieve.
use a SMC if you want to control your fans from cli. These values are used only in a few specific cases. And just settings fan mode and speed in defaults will not change your fan speed.