exelban/stats #1602: App doesn't display data in menu bar after Mac resumes from sleep

ubermlin, 2023-08-26:
**Describe the bug**

The stats app either quits or disappears from the menu bar of MacOS after I resume from sleep. I cannot replicate the issue, i.e. sometimes I can wake the computer several times and stats is there in the tray and sometimes it disappears after the first wake. 

I am not a proficient MacOS user (switched from Windows a short time ago), so I assume this might be something I am doing wrong. Still, decided to submit this.

**Details:**
 - Device: [M2 Mac Mini]
 - macOS: [Ventura 13.5.1 and prior]
 - Application version: [e.g. 2.9.6 and prior]

exelban, 2023-08-28:
Hi. Thanks for reporting. Could you please share the crash report?
It could be found in Applications -> Utilities -> Console. Find the Crash Reports tab. There you must see a row called Stats.

JanChrzan, 2023-08-28:
I have probably the same problem. After waking up the m2 max, the stats don't always show up. Sometimes the CPU disappears, sometimes the GPU disappears, practically 1/10 of attempts to wake up the computer end up with some statistics not appearing on the bar. It's helpful to use turn off and on in the app then everything reappears properly.

ubermlin, 2023-08-30:
> Hi. Thanks for reporting. Could you please share the crash report?
> It could be found in Applications -> Utilities -> Console. Find the Crash Reports tab. There you must see a row called Stats.

Hey, sorry it took so long. There's only one crash report in that tab and it's unrelated to Stats :(. Not sure how to help you troubleshoot this ;(.

exelban, 2023-08-30:
If there is no crash report, the app has no crash.
So the app closed after you wake up from sleep mode?
Also, what version of the app do you use?

ubermlin, 2023-08-30:
Hard to say if the app closed or crashed — if you say no report = no crash, you know better ;). For me as a user, it just disappears from the bar and and I need to open it again. Then the app stays in the bar until its gone again (after several sleeps and wakes). The mac wakes up with the app already gone (I never saw stats being present and disappearing from the bar right in front of me after a wake).

Currently I use 2.9.6 but this has happened on previous versions, too.

JanChrzan, 2023-08-30:
Now I know exactly what the problem is. At the time of system startup, applications start to run, which causes an excessive number of icons on the bar. When the barteneder starts first, this problem does not appear, because unnecessary icons jump into a separate bar, which causes the stats application to have space for its icons. However, if the order is reversed and the bartender starts last, the application does not show all the icons due to lack of space and even the bartener starts and moves unnecessary icons, the stats does not generate them again.

ubermlin, 2023-08-31:
I don't use bartender :(.

exelban, 2023-08-31:
it's two different not related cases

ubermlin, 2023-08-31:
I think so, too. Can I do anything to help you troubleshoot this?

exelban, 2023-08-31:
Unfortunately can't do anything without a crash report. And if Stats crashes, there must be a crash report. Otherwise, some other program kills the Stats.
Please monitor when Stats will be closed next time and check if there is a crash report.

ubermlin, 2023-09-04:
Hey!

[Stats-2023-09-04-085556.ips.zip](https://github.com/exelban/stats/files/12516983/Stats-2023-09-04-085556.ips.zip)

Managed to get a crash report. Let me know if this helps!

exelban, 2023-09-05:
yeap, it the same as in #1605 
trying to fix that(

closed 2023-09-05 by exelban