exelban/stats #1607: [Feature request] FPS widget in the GPU module

Amsitram, 2023-09-01:
There are currently 2 applications that can display FPS: iStat Menus and MenuBar Stats, but they are not open source.

exelban, 2023-09-02:
Heh, I want this feature from the beginning of the Stats. But cannot find a good way to receive the FPS from the system(
Maybe someone could help with that?

probabalyJackie, 2023-09-07:
> Heh, I want this feature from the beginning of the Stats. But cannot find a good way to receive the FPS from the system( Maybe someone could help with that?

Maybe with quartzz debug?

probabalyJackie, 2023-09-07:
maybe you can bundle it with your app

ivancuric, 2023-12-17:
I'd love to have this feature! One of the few I miss from iStatMenus!

Amsitram, 2024-03-25:
To detect FPS, MenuBar Stats apparently uses a companion application, an "SMC helper" (which doesn't require an admin password).

exelban, 2024-08-12:
hi. I'm going to close this ticket. I have tried a few times to implement that feature but failed. If someone knows the way to obtain FPS ping me please.

closed 2024-08-12 by exelban