exelban/stats #1610: [Issue]duplicate appliaction running

StevenLi-phoenix, 2023-09-04:
After enable open on start up, it open two stats at the same time. I believe it is my mistake that I downloaded the program to debug it and can not uninstall the debug version. Could you give me any suggestion on how to uninstall the debug version? Thank you.

 - Device: MacBook Pro 2020
 - macOS: V13.5.1 (22G90)
 - Application version: 2.9.6
[image]
[image]

exelban, 2023-09-06:
just delete the copy that you don't need.

closed 2023-09-06 by exelban

StevenLi-phoenix, 2023-09-18:
For anyone curious, the Xcode build is under path:
~/Library/Developer/Xcode/DerivedData/[build]
just delete that folder and reindex Spotlight Search