exelban/stats #1616: Suggestions for improving battery display

joonhyuk, 2023-09-15:
I've been happily replacing my Mac's default menu bar battery icon with stats to show charge and discharge times along with percentages. 

I'd like to suggest some minor improvements for better UX, #1 is a must-have, and #2 seems like it could be debated.
And since it's all about UX, so different people may react differently. The developer's opinion would be the most important I think.

#1. Charging status notification
There seems to be a widely accepted convention, like a lightning bolt on the battery icon.

#2. Better details (charge/discharge time)
Currently, when the charge/discharge time cannot be displayed, it says "n/a".
The value itself is 'not available' at that point, so it's correct, but it's not pretty.
Why not split it between fully charged (charger connected) and discharge time being calculated?

Suggestions:
- Charging and fully charged : If #1 is implemented, it would be nice to have an empty or "-" display.
- Calculating discharge time : "..." (blinking?)
- Rest of the exceptions (fallback) : "n/a"

Thanks,
JC

exelban, 2023-11-09:
Hi. Thanks for your feedback.

I take a look at what could be changed. And found no changes that I want to make in the Battery module.
Calculating time is more logical for me than `...`. Because `...` does not straightforward mean that time calculating.

About `n/a` - I think you are talking about the widget. You can hide that value if fully charged. And will not see n/a.

closed 2023-11-09 by exelban

austinarchibald, 2024-03-18:
+1 for having battery icon change when charging. Agree it's a must-have.

austinarchibald, 2024-08-03:
Just realized you can get the charging indicator when you use the Battery widget, but not the Bar chart. I like the minimalistic look of the Bar chart better. Is it possible to add the lightning bolt to the Bar chart when it's charging like the Battery one does? 

[image]

s-rog, 2024-12-05:
> Just realized you can get the charging indicator when you use the Battery widget, but not the Bar chart. I like the minimalistic look of the Bar chart better. Is it possible to add the lightning bolt to the Bar chart when it's charging like the Battery one does?
> 
> <img alt="image" width="80" src="https://private-user-images.githubusercontent.com/12240843/354839610-5b0fd3ee-4694-4d86-8933-67f6af546c75.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM0MjUyNDAsIm5iZiI6MTczMzQyNDk0MCwicGF0aCI6Ii8xMjI0MDg0My8zNTQ4Mzk2MTAtNWIwZmQzZWUtNDY5NC00ZDg2LTg5MzMtNjdmNmFmNTQ2Yzc1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjA1VDE4NTU0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE1MDRiN2Q4NWVhMDk5ZjY3ZmE0Yjk0NzRjNGY1ZmFmOTczMjk1ZjFiM2JlMGE3NThlYmE1ZTk1ZjE4Y2E4MjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.dzIqaa0GemOP4tkahMBBenBglRCvjq_xsAuPogVmzK4">

Would love this as well