exelban/stats #1621: System sensor icon on MacOS 12.6.8

msolodilin, 2023-09-22:
Hi! I really love this app bit there is one small issue: the icon for the system sensors appears as a restriction sign or something of that sort (see screenshot). I'm not sure if this is a bug or if I've done something wrong, but I thought it should be reported. Anyways, great app, thank you so much for making it open source!
[image]

exelban, 2023-09-22:
yes, you need to enable some sensors to show or disable the module/widget.

closed 2023-09-22 by exelban

msolodilin, 2023-10-02:
Thanks for the advice! If I understand the interface correctly, I have enabled several sensors (Fans, CPU temps, Wattage) as you can see on the screenshots below. Did I understand this correctly or does enabling mean something else?
[image]
[image]