exelban/stats #1623: [Feature Request] Right-click module for more details in Combined Mode

kevllarco, 2023-09-22:
Thanks for developing Stats! I love it. I like using it in Combined Mode as it's easier to manage and move around. The only problem is you lose a lot of the details in the dropdown menus. Is there a way you could allow users to ctrl-click or right-click to get access to the full module detail?

For example, while in Combined Mode, the user would have to ctrl-click or option-click to get the full details of the CPU module and see all the processes running. I think a direct keyboard shortcut for each module would resolve this issue also but I didn't see any option for that. Thanks so much!

justinharrell, 2023-10-01:
Definitely second this seems duplicate of #1604, #1497, #1353, #1449 and probably more. 

Seems like it should be relatively simple to display the separated popup next to the combined popup. iStat does this on hover on the combined module, so if you mouse over cpu the separated cpu pane pops next to it. Hover can be a little aggressive I would be fine with just a single click popping the separated details up and of course being able to drag to keep it open.

The other issue with combined is not being able to hide widgets while still seeing them in the popup, that is on the menu bar I just want to see battery cpu and network, but if I click on it then see all the other modules.

closed 2023-10-05 by exelban

mailinglists35, 2024-01-24:
hi @exelban 
why is this not planned? is it technically impossible?