exelban/stats #1624: Disk reader - menubar widget shows incorrect value after some time

luckman212, 2023-09-22:
Yesterday I noticed my disk space got up to 96% consumed, but I didn't realize because Stats menubar widget had ceased to update. Here's a screenshot as it was filling.

[image]

_(This is from 13.5.2 + Stats 2.9.7, but since then I have updated to 13.6)_

I don't know exactly the circumstances that cause this, but the space was being consumed by Time Machine local snapshots (I don't use Time Machine but on APFS disks you can't turn these local snapshots off). Some kind of runaway process there. I cleared them with the following terminal commands

```bash
tmutil deletelocalsnapshots /
tmutil thinlocalsnapshots / 1000000000000 1
```

So again not sure if it's a bug in macOS or in Stats.

Related issues

- https://github.com/exelban/stats/issues/1470
- https://github.com/exelban/stats/issues/1510
- https://github.com/exelban/stats/issues/1552
- https://github.com/exelban/stats/issues/1589

Thank you for looking 🙏

exelban, 2023-09-22:
why do you mark these issues as related?
all users receive a notification of that. And it is nothing related to these issues.

Also please share the screenshot of disk module settings.

luckman212, 2023-09-22:
Sorry, I thought they might be related. I removed that. 😔

Here are screenshots of my disk module settings.

[image]

[image]

[image]

exelban, 2023-09-22:
thx, nope. The problem that was in this issue was related to a bug in macOS. And it's fixed already. The free space does not include the purgable space.

Does the Stats restart help?
Or even after the restart widget shows a different value than the popup?

luckman212, 2023-09-22:
I am sorry but I don't remember if restarting Stats made the widget correctly display. But, I am 80% sure it still had the wrong value.

Next time (if there is a next time) I will make some more observations. When you say it was a macOS bug that's already fixed, do you mean in 13.6 or 14.0?

exelban, 2023-09-27:
Hi, did you see this behavior again?

exelban, 2023-09-27:
also, few questions. Is it your main (root) disk on the mac ('mini')?
Did you change the disk name last time?

luckman212, 2023-09-27:
Yes the main root disk is named "mini". It has been named that since I got the computer.

I have not seen the problem again (yet) but I also just installed 14.0 Sonoma, so maybe this is a 13.x bug. I definitely think that was the buggiest OS I used in a long time.

exelban, 2023-09-27:
hmm, because I don't see an option to not update the widget until the disk was manually selected, and after the name was changed. Because there is name mapping(
Please ping me if it repeats, and I will send you a debug build.

exelban, 2023-09-27:
@luckman212 just curious.
In the first screenshot widget shows ~25%, and the popup 92%. What value is close to the system value?

luckman212, 2023-09-27:
In that case, it was 92%. But the "real" usage was around 55%, the rest was being eaten up by local APFS snapshots which had gotten out of control (bug). So it was probably marked as purgeable space, maybe the reason that the readers were giving different values.

But yes, the "real" value was 92% and I was starting to get alarms from other software on my computer saying disk almost full, etc. @exelban

exelban, 2023-09-27:
What do you mean by "real"? Do you check the settings panel with storage at that time maybe?
Stats uses the API that gets the free space including the purgable. Because apple uses the same. And it's what users expect to see (me too).

Playing with the new macOS and seeing strange behavior. Free space does not change after deleting 50+ GB of data. Even settings panel does not shows the changes.
Only after sleep -> wake up the space is free.

exelban, 2023-09-27:
yeap. there is a bug while updating the disk view in the popup(
So the widget shows the correct value. But the popup view shows the old one.

luckman212, 2023-09-27:
Well yes I agree with APFS the whole thing is very mysterious. Many layers of obfuscation trying to hide what's going on from the user.

When I say "real" I mean, what does Finder → Disk → Get Info show, and also what is reported by DaisyDisk and other such apps...

[image]

The only reason I even knew my disk was full was because I use [Syncthing](https://github.com/syncthing/syncthing-macos/) and it is set to throw an alarm and stop syncing when the disk reaches 95% full. Otherwise, just looking at Stats menubar widget I would not have suspected anything. I was not even actively using the computer.

closed 2023-10-01 by exelban

luckman212, 2023-10-01:
Thanks @exelban ! 🎉