exelban/stats #1626: [Feature Request] Desktop Widget

TeackReais, 2023-09-24:
Apple recently released macOS 14 Sonoma, which brings a major upgrade to the desktop widgets functionality. With macOS 14, users can place widgets on their desktop from the new widget gallery. I wish the Stats app would support widgets, which would be easier to view on the desktop than in the menu bar. Thanks.

exelban, 2023-09-25:
macOS 14 is not released yet.

Sayor0630, 2023-09-27:
> macOS 14 is not released yet.

It is released.

exelban, 2023-09-27:
yeap, now it's released

darkguy2008, 2023-10-05:
I just came here through a google search looking for this lol. It'd be awesome to have some desktop widgets that have the same info of the popups when you click one of the stats item. Hope to see this someday soon! :D

exelban, 2023-10-21:
it's not possible right now. Due to some bugs in the Xcode I cannot add a widget kit target to the project. Will add when it will be possible.

exelban, 2023-11-06:
will leave that ticket open to avoid duplicating.

exelban, 2023-12-23:
Ok, I finally figured out what caused the problem to add the Widget extension to the project.
For some reason widget extension cannot added to this project without changing the bundle identifier. I have no idea why but changing even 1 letter in the current bundle identifier makes the widget extension work fine. But in the current state, it cannot be added.

So there are only 2 options: 
- to change the app bundle identifier
- not implementing widgets

I was thinking a long ago about changing the bundle identifier. So I will try to do that. If it cannot be done, it will be impossible to add the widgets support to the project. Or maybe Apple will fix that problem.

acicovic, 2024-06-07:
Hello, I just wanted to +1 this idea 🙂

In my case, moving stats to a desktop widget would allow me to see all my menu bar items. Currently, some of them are hidden under the notch and the stats take a lot of place.

cosmic-latte, 2024-06-30:
I agree. It would be great to have a desktop widget that shows RAM or CPU usage separately.

exelban, 2024-08-15:
Hi. I have added a widget for the next modules: CPU, GPU, RAM, Disk, Network.
For now there will be widgets only for that modules. I will add another with time. But for sure there will be no widget for sensors since I have no idea of how so much data could be represented in the widget.

closed 2024-08-15 by exelban