exelban/stats #1632: After upgrade to Sonoma Network widget not showing the download values in the widget

ricardonadao, 2023-09-29:
After the upgrade to Sonoma, stats 2.9.7 stop showing the download numbers in the widget when using "interface based"

[image]

if we move to "process based" works:

[image]

**Details:**
 - Device: Mac Book Pro M1
 - macOS: 14.0
 - Application version: 2.9.7

Noksa, 2023-09-29:
Encounter the same on my Macbook Pro M1

Noksa, 2023-10-04:
@exelban Hello.
Is there any ETA when it will be fixed?

exelban, 2023-10-05:
@Noksa nope. Because it's not the problem in the Stats. The only option in your case it's to switch to the process based reader.

zzJinux, 2023-10-06:
Hi, @exelban
> Because it's not the problem in the Stats.
You're saying that the interface based reader does work on Sonoma and the author's problem is related to something else?

exelban, 2023-10-13:
[Stats.dmg.zip](https://github.com/exelban/stats/files/12892055/Stats.dmg.zip)

Please run this build for a few minutes and share the logs that could be found in `Documents/logs.txt`

exelban, 2023-10-13:
@ricardonadao do you have any download traffic in that time?

ricardonadao, 2023-10-13:
Yes, I was doing normal things, but I can run it again and start a big
download if you prefer?

ricardonadao, 2023-10-13:
[log.txt](https://github.com/exelban/stats/files/12894308/log.txt)

Done with some big downloads.

exelban, 2023-10-13:
thx. Unfortunately, I cannot help with that problem. The download value (last in the row) is not changing as you can see in the logs. It means the driver does not report the correct values. The only option for now is to use processes as I/O source.

Interesting bug, because I don't have this problem on any of my macs with macOS sonoma.

ricardonadao, 2023-10-13:
I have an Intel and a M1 Mac book pro both with the same issue.

ricardonadao, 2023-10-16:
is there a way to reset the counters?

gs-arthur, 2023-10-18:
Having a similar issue on my M1 Mac. Happened with the plugin Bandwidth+ as well

exelban, 2023-10-21:
going to close this issue, because it's not a problem in the Stats. If you have this issue use a processes based reader.

closed 2023-10-21 by exelban

ricardonadao, 2023-11-15:
suddenly it started working for me again since yesterday in one of the Macs, need to check the other one.

Rhyalus, 2023-12-15:
M2 MB Air - latest OS - same issue.  The download speeds are shown if you click on the menu bar item (under processes).  Changing to process based instead of interfaced base "fixes" the problem.  Is there a downside to setting it to process based?

R

grifosh, 2024-08-20:
Having a similar issue on my M3 Mac.

ricardonadao, 2025-01-06:
Still with this issue, in one of my Macs it endup solving it by itself.

In the M1 Mac it only seems to count the "upload" when setup to render interface info, need to move it to process based to work.