exelban/stats #1638: Stats 2.9.8 crashes immediately with SIGABRT in Thread 3 - com.apple.root.background-qos

luckman212, 2023-10-02:
woke up my Mac this morning to find this dialog

[image]

never had this happen before

relaunching, it just crashes immediately.

guessing this is a macOS Sonoma bug.

stack trace:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [46406]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.9.8 (501)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-10-01 13:51:47.9247 -0400
OS Version:            macOS 14.0 (23A344)
Report Version:        12
Anonymous UUID:        A6DF521A-3508-6D9A-A2C2-BF0317E32BFC

Time Awake Since Boot: 92000 seconds

System Integrity Protection: enabled

Crashed Thread:        3  Dispatch queue: com.apple.root.background-qos

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   Stats [46406]

Application Specific Information:
stack buffer overflow

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x18c04e8b4 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x18c060d30 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x18c0574f0 mach_msg_overwrite + 476
3   libsystem_kernel.dylib        	       0x18c04ec38 mach_msg + 24
4   CoreFoundation                	       0x18c16bed0 __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x18c16a798 __CFRunLoopRun + 1208
6   CoreFoundation                	       0x18c169c2c CFRunLoopRunSpecific + 608
7   HIToolbox                     	       0x1966c2448 RunCurrentEventLoopInMode + 292
8   HIToolbox                     	       0x1966c2284 ReceiveNextEventCommon + 648
9   HIToolbox                     	       0x1966c1fdc _BlockUntilNextEventMatchingListInModeWithFilter + 76
10  AppKit                        	       0x18f944f90 _DPSNextEvent + 660
11  AppKit                        	       0x190118b94 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716
12  AppKit                        	       0x18f93843c -[NSApplication run] + 476
13  Stats                         	       0x100946a54 0x100944000 + 10836
14  dyld                          	       0x18bd15058 start + 2224

Thread 1::  Dispatch queue: com.apple.root.background-qos
0   libicucore.A.dylib            	       0x18f2271ac icu::UnicodeSet::applyFilter(signed char (*)(int, void*), void*, icu::UnicodeSet const*, UErrorCode&) + 180
1   libicucore.A.dylib            	       0x18f2272e4 icu::UnicodeSet::applyIntPropertyValue(UProperty, int, UErrorCode&) + 180
2   libicucore.A.dylib            	       0x18f33c980 icu::RegexCompile::matchStartType() + 924
3   libicucore.A.dylib            	       0x18f339730 icu::RegexCompile::compile(UText*, UParseError&, UErrorCode&) + 652
4   libicucore.A.dylib            	       0x18f35436c icu::RegexPattern::compile(UText*, unsigned int, UParseError&, UErrorCode&) + 448
5   libicucore.A.dylib            	       0x18f3a83f8 uregex_open + 360
6   Foundation                    	       0x18d26db7c -[NSRegularExpression initWithPattern:options:error:] + 308
[... 98 lines cut ...]
       0x100ba0000 -        0x100babfff libobjc-trampolines.dylib (*) <562f95b3-8118-3d61-a13f-34e819dd863d> /usr/lib/libobjc-trampolines.dylib
       0x100e88000 -        0x100eb7fff eu.exelban.Stats.Disk (1.0) <2bc87748-7ded-3bc3-9c39-1b07b5f89d39> /Applications/Stats.app/Contents/Frameworks/Disk.framework/Versions/A/Disk
       0x100e40000 -        0x100e57fff eu.exelban.Stats.Clock (1.0) <067f4042-c110-3724-8ed3-d9fb065e50ff> /Applications/Stats.app/Contents/Frameworks/Clock.framework/Versions/A/Clock
       0x100ddc000 -        0x100e1bfff eu.exelban.Stats.Sensors (1.0) <ddb986c9-33c1-39a6-9f1c-3aae9c2c07c9> /Applications/Stats.app/Contents/Frameworks/Sensors.framework/Versions/A/Sensors
       0x100f44000 -        0x100f5bfff eu.exelban.Stats.Battery (1.0) <76ad0df8-7dbf-3265-b89a-d356d03afa1c> /Applications/Stats.app/Contents/Frameworks/Battery.framework/Versions/A/Battery
       0x100edc000 -        0x100ef7fff eu.exelban.Stats.Bluetooth (1.0) <0fec08b1-f7f8-305a-b70a-d4c3969b0bac> /Applications/Stats.app/Contents/Frameworks/Bluetooth.framework/Versions/A/Bluetooth
       0x101194000 -        0x10124bfff eu.exelban.Stats.Kit (1.0) <18654d34-f018-3454-bba3-c72cd0c106b8> /Applications/Stats.app/Contents/Frameworks/Kit.framework/Versions/A/Kit
       0x100fe8000 -        0x101003fff eu.exelban.Stats.RAM (1.0) <a2fbd622-829a-3d32-a92a-217620687f40> /Applications/Stats.app/Contents/Frameworks/RAM.framework/Versions/A/RAM
       0x100f74000 -        0x100faffff eu.exelban.Stats.Net (1.0) <07882831-e45e-310d-9321-d4bec7ff8307> /Applications/Stats.app/Contents/Frameworks/Net.framework/Versions/A/Net
       0x1010a8000 -        0x1010cbfff eu.exelban.Stats.GPU (1.0) <dbebe472-3134-3afa-bf22-17c0388c06da> /Applications/Stats.app/Contents/Frameworks/GPU.framework/Versions/A/GPU
       0x101020000 -        0x101043fff eu.exelban.Stats.CPU (1.0) <bfadf5bf-fc70-3868-abf2-f8e26f8b26a7> /Applications/Stats.app/Contents/Frameworks/CPU.framework/Versions/A/CPU
       0x100944000 -        0x100973fff eu.exelban.Stats (2.9.8) <9b0405b7-21cf-32bb-bbd2-cb433fc8db4c> /Applications/Stats.app/Contents/MacOS/Stats
       0x18c04d000 -        0x18c087fef libsystem_kernel.dylib (*) <a7d3c07d-0a1e-3c4c-8fba-66905e16bf99> /usr/lib/system/libsystem_kernel.dylib
       0x18c0ee000 -        0x18c5c4fff com.apple.CoreFoundation (6.9) <a68b8c77-1dbd-35b0-83fe-42ad58dd6629> /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
       0x196692000 -        0x196951fff com.apple.HIToolbox (2.1.1) <38ee0ac5-f6f6-391c-852e-6954b5d253e8> /System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox
       0x18f90b000 -        0x190c1bfff com.apple.AppKit (6.9) <a4e78dd1-6b6e-3f57-924a-4a6a2e679789> /System/Library/Frameworks/AppKit.framework/Versions/C/AppKit
       0x18bd0f000 -        0x18bda2873 dyld (*) <ffd8ab66-c9ab-31df-ab80-3a3dff367ddd> /usr/lib/dyld
               0x0 - 0xffffffffffffffff ??? (*) <00000000-0000-0000-0000-000000000000> ???
       0x18f178000 -        0x18f430ff7 libicucore.A.dylib (*) <bd4eb2c6-787c-39f4-be9a-170def3dbe70> /usr/lib/libicucore.A.dylib
       0x18d217000 -        0x18de69fff com.apple.Foundation (6.9) <11eb37ae-355b-3a35-af1b-13b599244410> /System/Library/Frameworks/Foundation.framework/Versions/C/Foundation
       0x18bede000 -        0x18bf24fff libdispatch.dylib (*) <8790ba20-19ec-3a36-8975-e34382d9747c> /usr/lib/system/libdispatch.dylib
       0x18c088000 -        0x18c094ff3 libsystem_pthread.dylib (*) <e4debb6e-421d-33d0-9e17-77ae0e0fe4dc> /usr/lib/system/libsystem_pthread.dylib
       0x18c0ba000 -        0x18c0c0ffb libsystem_platform.dylib (*) <e7e2f10a-772a-397f-bd19-a7ea5a54e49f> /usr/lib/system/libsystem_platform.dylib
       0x18bf28000 -        0x18bfa6ff3 libsystem_c.dylib (*) <1b84a7e4-8958-330c-98b8-27d491dff69e> /usr/lib/system/libsystem_c.dylib
```

luckman212, 2023-10-02:
so weird, I tried maybe 7 times to relaunch Stats - crashed every time. I even logged off and logged back on to clear my session.

Then, I downloaded 2.9.7 and ran it to test, it did not crash.

Then I re-downloaded 2.9.8 and ran THAT again, and it also did not crash. 🤷

exelban, 2023-10-02:
Please re-download the release manually. I updated the release build some time after release when I discovered that problem.

closed 2023-10-02 by exelban