exelban/stats #1659: Crash when disk widget is enabled

geiseri, 2023-10-17:
This may be another character in the saga of #1520 , but maybe the crash reports will give some more info.  They all end in a subtly different place, but they do all seem to cross through trying to draw a composited bitmap.  They all seem to be happening when `volumePropertyProviderPrepareValues` along with cache eviction.  I disabled the widget and will see if that might be the source of troubles.

I am on a 2018 macmini so maybe my system is slow enough that it aggravates the race somehow?

I took a subset, but they all seem to have the same general idea:

[Stats-2023-10-10-115920.ips.txt](https://github.com/exelban/stats/files/12937367/Stats-2023-10-10-115920.ips.txt)
[Stats-2023-10-11-154647.ips.txt](https://github.com/exelban/stats/files/12937368/Stats-2023-10-11-154647.ips.txt)
[Stats-2023-10-12-142237.ips.txt](https://github.com/exelban/stats/files/12937369/Stats-2023-10-12-142237.ips.txt)
[Stats-2023-10-13-004424.ips.txt](https://github.com/exelban/stats/files/12937370/Stats-2023-10-13-004424.ips.txt)
[Stats-2023-10-14-162937.ips.txt](https://github.com/exelban/stats/files/12937371/Stats-2023-10-14-162937.ips.txt)
[Stats-2023-10-16-175832.ips.txt](https://github.com/exelban/stats/files/12937372/Stats-2023-10-16-175832.ips.txt)

exelban, 2023-10-17:
hi. Thanks for the report.
Looks like it's a new bug. The problem is not in the Disk module, but in the stack widget (according to the crash reports).

exelban, 2023-10-17:
Please check all stack widgets if you have the monospaced option enabled?

geiseri, 2023-10-19:
Monchrome or is it a specific widget type: 
[image]

Maybe I am blind but I didn't see monospace as an option.

geiseri, 2023-10-27:
I didn't have those enabled.  That being said I reset all the settings to default and it has not crashed since last week.  I am going to update today and enable one thing at a time, hopefully that will help narrow it down.

exelban, 2023-11-09:
Closing this issue. Ping me if in case of new information.

closed 2023-11-09 by exelban