exelban/stats #1661: I don't see sensors on M1

denisborisov, 2023-10-20:
Hello.

I installed Stats in my Apple M1 Pro and even after enabling `HID sensors` option I do not see any sensors.

My macOS version is 13.5.2 (22G91)

nikdpred, 2023-10-21:
I've been sitting on one of the old versions for a year and everything was great. Then I accidentally updated Stats a month ago and unfortunately it became completely useless software. CPU module never appears in the bar when the system boots, if I add it manually it only appears after restarting Stats app. Disk modules shows some random old numbers and never updates in real time. Sensors may or may not work at random. None of the recent updates have fixed any of the issues.  13.4.1 (22F82)

closed 2023-10-21 by exelban

denisborisov, 2023-10-21:
@exelban , could you explain a little bit your decision to close the ticket? Readme says that in case of a problem one should open an issue.

sumitsahoo, 2023-10-22:
@exelban I am also not able to see any sensor data in menubar on my M1 Pro MacBook Pro. I am on Sonoma (macOS 14.0). Screenshot below:

[image]

denisborisov, 2023-10-22:
I don’t see anything in menubar. :(