exelban/stats #1662: [Feature request] Display the power of the charger

Vestrond, 2023-10-20:
It would be really useful if the app would display a charger power information such as "100W" or "35W" when the charger is connected. Now this information is available only at "Settings -> General -> About -> System Report -> Hardware -> Power -> AC Charger Information -> Wattage (W)".

Thank you!

exelban, 2023-10-22:
you can find this info in the popup view.

closed 2023-10-22 by exelban

Vestrond, 2023-10-23:
Oh, now I see. That's why I haven't seen it yet:
1. Unfortunately, there is no access to this parameter in Combined Mode.
[image]
2. This information cannot be displayed on the dashboard in any way (no such params in the options).

But I see what you're getting at. You want Combined mode to be minimised. So I am not sure if it is acceptable to add Wattage to the Combined panel.

Honestly, it looks something like this in my head:
[image]

or even:
[image]

So if you do something like this, it would be very handy. Thanks for the cool app!