exelban/stats #1667: CPU % in top processes is out of bound by long process name

wtchangdm, 2023-10-26:
**Describe the bug**
Hi, first of all, thanks for the this awesome tool!

I recently noticed that the CPU % in top processes became out of bound because of the relatively long process name like this:

[image]

I think it could be reproduced by changing a arbitrary binary to very long name.

Please let me know if further information is needed. Thanks!

**Details:**
 - Device: Macbook Pro 2023
 - macOS: 14.1
 - Application version: 2.9.10

rajaryan12123, 2023-10-28:
However long the name is, I think the percentage number should still be visible with the current implementation of processView() 🤔

I tried keeping the process name to be very long, but still the percentage seems to be present. Help me reproduce this bug.

[image]

wtchangdm, 2023-10-29:
Hmm, I think another requirement here is to get over 100% per process.

A simple way to reproduce:

```go
# veryveryveryveryveryveryveryveryveryverylong.go
package main

func main() {
	hold := make(chan interface{})

	for i := 0; i < 2; i++ {
		go func () {
			for {}
		} ()
	}

	<- hold
}
```

Then `$ go run ./veryveryveryveryveryveryveryveryveryverylong.go`

[image]

exelban, 2023-10-29:
@wtchangdm thanks for that. I have been trying to reproduce that bug for a long time but cannot find a reason why it's presenting. In my case, it is always with the docker process.

Will be fixed.

exelban, 2023-10-29:
it does not work for me, for some reason(

[image]

wtchangdm, 2023-10-29:
@exelban 
Thanks for the fast reply! I tried with two processes that are > 100% but still observed the same issue.

(Not sure why the screenshot is so huge than OP, I will just leave a link here)
https://i.imgur.com/eAxyZxv.png

closed 2023-11-05 by exelban