exelban/stats #1673: [Feature Request] Ability to add GPU, CPU, and battery sensor data in those widgets

noel-yap, 2023-11-01:
I'd like to group relevant sensor data (eg temperature) with the GPU, CPU, and battery widgets. Can this option be added?

exelban, 2023-11-05:
I don't understand what you want to achieve. What is the problem with grouping the sensors in the widget in the way you want?

noel-yap, 2023-11-05:
Currently all sensor widgets must be grouped together. I would like to
group GPU-related sensor data (eg GPU temperature) with the other GPU
widgets, CPU-related sensor data with the other CPU widgets, and
battery-related sensor data with the other battery widgets.

noel-yap, 2023-11-05:
Attached is a snapshot of my current widget arrangement.

That 51° is the hottest CPU while the 49° is the hottest GPU and the two
31° are the battery temperatures. I would like to place that 51° with the
CPU widgets, the 49° with the GPU widgets, etc.

exelban, 2023-11-09:
Sensors are available only in the Sensors module. You can use any widget in this module to arrange the items in the menu bar.

closed 2023-11-09 by exelban

noel-yap, 2023-11-09:
I understand that's how it is today. I'm requesting that sensors to be made
available in the modules that relate to them (eg CPU sensors to be made
available in the CPU module). In this way, all, say, CPU-related widgets
can be merged as one widget and moved around as one. I've also not found a
way to split the sensor module into separate widgets such that, say,
CPU-related widgets can be placed next the CPU module.