exelban/stats #1682: How to uninstall Stats?

xinglinquan, 2023-11-07:
Hello, may I ask how to completely uninstall Stats? What I mean is that after Stats is uninstalled, the relevant settings, modifications, or generated files that Stats has changed will be restored. 
PS: I installed it manually.

F1248, 2023-11-07:
You can use free utilities like [AppCleaner](https://freemacsoft.net/appcleaner/).

exelban, 2023-11-07:
normally it's enough to delete the app from the Application folder. Stats do not store the data. 
Only if you used fan control you need to uninstall helper which could be done from the app settings.
If not, removing the app from the Application will be ok.

closed 2023-11-07 by exelban

xinglinquan, 2023-11-08:
Thank you very much, so I would like to confirm that if I uninstall the fan control program in the Stats settings, will the fan control program be uninstalled completely, that is, will the relevant system settings modified or generated by the fan control program be restored?
Finally, when I input `launchctl print disabled system` on the terminal, I see the following output, as shown in the figure below. I would like to know if the content in the red box of the penultimate item `"fans"=>enabled`  is related to Stats or is it generated by Stats? To my knowledge, macbookpro generally does not have this service option.
[image]

exelban, 2023-11-08:
I have no idea what the service is for. 
For sure not for Stats.

xinglinquan, 2023-11-09:
Thanks, let's go back to the previous question.  If I uninstall the fan control program in the Stats settings, will the fan control program be uninstalled completely, that is, will the relevant system settings modified or generated by the fan control program be restored?

exelban, 2023-11-09:
I do not understand the question. If you don't want to control the fans by Stats just uninstall the Stats helper from the settings. Thats all. After that you can delete the Stats from the application folder.

xinglinquan, 2023-11-09:
Sorry, I'm just worried that there will still be residual settings that haven't been restored after uninstalling the fan control program, which will affect the computer. That's what I want to confirm.