exelban/stats #1688: Manually adjusting the fan doesn't work.

DevLiuSir, 2023-11-13:
My device is m1 max 12.7.1。

[image]

akhoury6, 2023-11-17:
I am seeing the same issue on a 2023 16" M3 Max. The controls are enabled but I can not make any adjustments.

DevLiuSir, 2023-11-17:
I see many of the closed issues of the same person.

DevLiuSir, 2023-11-17:
> I am seeing the same issue on a 2023 16" M3 Max. The controls are enabled but I can not make any adjustments.

I see many of the same problems that have been closed.

da-vinci-noob, 2023-11-18:
Fan speed adjustments break on Sonama upgrade, was working fine with Ventura.

DevLiuSir, 2023-11-18:
> Fan speed adjustments break on Sonama upgrade, was working fine with Ventura.

I tried many times under Monterey to uninstall the app and reinstall the fan assistant, and the same result was that I couldn't adjust the fan speed.

da-vinci-noob, 2023-11-18:
> > Fan speed adjustments break on Sonama upgrade, was working fine with Ventura.
> 
> I tried many times under Monterey to uninstall the app and reinstall the fan assistant, and the same result was that I couldn't adjust the fan speed.

Can you please show the screenshot of "Allow in the Background" section at "Login Items" Page

eugkhp, 2023-11-24:
Same for me, after recent macos update the fans are not adjustable

DevLiuSir, 2023-11-28:
Use the compiled smc binary file and use the sudo command to operate the fan. But the button adjustment has no effect.

joujek, 2023-11-30:
same for me M2 Pro 14 12/19

xiaozealer, 2023-12-04:
same for me M3 max 16 inch

gkndr, 2023-12-10:
same issue for M1 Pro with Sonoma

turannul, 2023-12-13:
[image]
I also encountered this issue, fix was enabling launchd/login item in `Settings - General - Login Items`

turannul, 2023-12-13:
> <img alt="Screenshot 2023-12-13 at 20 14 19" width="502" src="https://private-user-images.githubusercontent.com/84424530/290273320-f917b0bf-7e8f-4092-848f-8ae072563c14.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI0ODgyNjQsIm5iZiI6MTcwMjQ4Nzk2NCwicGF0aCI6Ii84NDQyNDUzMC8yOTAyNzMzMjAtZjkxN2IwYmYtN2U4Zi00MDkyLTg0OGYtOGFlMDcyNTYzYzE0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjEzVDE3MTkyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZiMTBkMjdkNGJhNzMzMjRiN2FiMjg0NWQ2Yzc1NDYzNTZhM2ZiOTQyNmU2ZTk3ZDQxYmM5OTY2YWE0MzZlZjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.b_ZJUYGRs7uOXl8iecmc15Il5P5nckmwDkpwa1MUFFQ"> I also encountered this particular, fix was enabling launchd/login item in Settings - General - Login Items

I'd like to point out, 
The connection between this "(Login item name Serhiy Mytrovtsiy)" and Stats app is not easily understandable without a quick Google search.

joujek, 2023-12-13:
Confirmed - works for me after enabled Item in Login Items -> "Allow in the Background" section.

M2 Pro 14/19- Mac14,9

xiaozealer, 2023-12-13:
Still doesn't work for me even with " Serhiy Mytrovtsiy" enabled M3 Max 16inch 
[image]

joujek, 2023-12-13:
hi @xiaozealer - Did you restarted Stats app after enable it in "Allow in the Background" section

xiaozealer, 2023-12-13:
> hi @xiaozealer - Did you restarted Stats app after enable it in "Allow in the Background" section

Yes, I did. I tried two times.

exelban, 2023-12-13:
Could you please check if you have the `eu.exelban.Stats.SMC.Helper` process running?

xiaozealer, 2023-12-13:
@exelban yes, it is running
[image]

exelban, 2023-12-13:
please try to change the fan speed monitoring at the same time that process. Does it restart when you trying to control the fan?

xiaozealer, 2023-12-13:
> please try to change the fan speed monitoring at the same time that process. Does it restart when you trying to control the fan?

no it doesn't restart the process, it stays.

exelban, 2023-12-13:
strange. Looks like the problem is not in the Stats. Ok, will try to make a debug build with a special debugger.

xiaozealer, 2023-12-15:
@exelban I discovered an intriguing method to manually activate the cooling fan of my computer. Initially, I run some benchmark tests, which naturally caused the system to engage the fan for cooling. Intriguingly, during this process, I was able to manually manipulate the fan speed. However, I noticed a limitation to this manual control. Once the computer reached an adequate cooling level and the fan stopped operation, I found that I could no longer manually initiate the fan again. 
[image]

exelban, 2023-12-15:
do you maybe have some other tool that can control the fans?

xiaozealer, 2023-12-15:
no, stat is the only software I installed that is able to control the fans

exelban, 2023-12-17:
hi. Please try this build:

[Stats.dmg.zip](https://github.com/exelban/stats/files/13696223/Stats.dmg.zip)

xiaozealer, 2023-12-17:
@exelban seems like still not working
[image]

exelban, 2023-12-17:
ok, so please do the following:
- run this build
- open the Console app (could be found in the utilities folder)
- enter `Stats` in the search field and select `Process`
- change some fan from auto to manual mode and try to change the speed

And please share in some way logs from the Console app. It will be better if you disable the Disk module in the Stats before doing that because it makes a lot of logs.

xiaozealer, 2023-12-17:
I tried switching between manual and automatic as well as changing some speed, below is the screenshot of the log
[image]

exelban, 2023-12-18:
There are no logs(
Please filter by `eu.exelban.Stats.SMC.Helper` and try to adjust a fan speed/mode.

xiaozealer, 2023-12-18:
I was not able to trigger eu.exelban.Stats.SMC.Helper logs to appear despite I restarted the app multiple times and do speed change, so I filtered by SMC only and have the following 
[image]

exelban, 2023-12-18:
nope, please use only the `eu.exelban.Stats.SMC.Helper` filter.
Ok, so please uninstall and install again the helpers and try again.

xiaozealer, 2023-12-18:
After reinstallation, it says ok but the fan speed doesn't change, weird 
[image]
[image]

xiaozealer, 2023-12-18:
[image]

exelban, 2023-12-18:
hm, so the fans are in the forced mode. They don't want to change the target speed...

exelban, 2023-12-18:
did you try maybe manually changing the target speed?

xiaozealer, 2023-12-18:
[image]

exelban, 2023-12-18:
please comment what did you do?)

Do you manually change the target speed or with Stats?
do the fans change their speed?

xiaozealer, 2023-12-18:
so I set the mode in Stats to Manual and set the speed to both fans as you can see from the screenshot, but the fans aren't running.

exelban, 2023-12-18:
hmm
do you try to control the fans with any other tool?

xiaozealer, 2023-12-18:
I don't think I'm running other tools that is able to control the fans😂

exelban, 2023-12-18:
nope, I'm just asking if you have tried. Because looks like the problem is not in the Stats.

according to the last screenshot everything is ok:
- fan mode is manual
- and target speed is properly changed

so all that needs to change the fan speed is done.

xiaozealer, 2023-12-18:
Yeah... I know. It just doesn't work on my computer, not sure some other people have the same problem.

exelban, 2023-12-18:
There were a few reports like this.
In a lot of cases users block Stats helpers or looks like in your case it just does not work(

but still, it's very strange. Could you maybe try some other tool just to ensure that the problem is not in the Stats?

xiaozealer, 2023-12-18:
will try other fan control tools and let you know!

exelban, 2023-12-18:
thx. Will be waiting for feedback.

xiaozealer, 2023-12-18:
just tried TG Pro and Mac Fan Control, none of them worked. I guess it's my computer's problem. I'm going to reset SMC... thanks for you help!!

exelban, 2023-12-18:
hmm, share please the output of `/Applications/Stats.app/Contents/Resources/smc list -f`

xiaozealer, 2023-12-18:
```
[INFO]: found 30 keys

[F0Ac]     0.0
[F0CR]     38400.0
[F0Dc]     0.1599999964237213
[F0Fb]     8.0
[F0Fc]     65535.0
[F0Md]     0.0
[F0Mn]     1350.0
[F0Mx]     5349.0
[F0Sf]     0.0
[F0St]     1.0
[F0Tg]     0.0
[F1Ac]     0.0
[F1CR]     38400.0
[F1Dc]     0.1599999964237213
[F1Fb]     8.0
[F1Fc]     65535.0
[F1Md]     0.0
[F1Mn]     1458.0
[F1Mx]     5777.0
[F1Sf]     0.0
[F1St]     1.0
[F1Tg]     0.0
[FBAC]     0.0
[FBAD]     0.0
[FNum]     2.0
[FOFC]     14.0
[FOff]     1.0
[FRmp]     0.0
[Fpds]     0.0
[Frqd]     0.0
```

exelban, 2023-12-18:
yeap, looks like on some m3 (and looks like some m2 too) based Macs the fans cannot be controlled(

xiaozealer, 2023-12-18:
thanks @exelban.

closed 2023-12-24 by exelban