exelban/stats #1698: CPU Temperature being reported as 6C

TolleyLikesRice, 2023-11-24:
**Describe the bug**
My CPU Temperature is being (I assume) incorrectly reported as 6C. Under heavy load, the temperature does appear to read correctly around 90C

**Details:**
 - Device: 2023 14" Macbook Pro
 - macOS: 14.1
 - Application version: 2.9.11

exelban, 2023-11-24:
Hi. What CPU do you have?

TolleyLikesRice, 2023-11-25:
Hi, it's an M2 Pro 12/18

greathch, 2023-11-27:
Me too, M2 pro

joujek, 2023-11-27:
Same for me 12c/19g
[image]

In other hand on the sensors tab I can see temperatures a big higher than other similar tools are reporting - eg. iStatsMenu or Macs fan speed:

[image]

I suppose it could be caused by : https://github.com/exelban/stats/issues/1695

joujek, 2023-12-12:
Dear @exelban 
Still the same issue on M2Pro - Temp sometimes is correct but frequently goes to 6C:
[image]

exelban, 2023-12-12:
@joujek hi. Hmm, it's strange. Could you share the sensors popup view?

joujek, 2023-12-12:
HI @exelban screensots below:
[image]
[image]

joujek, 2023-12-13:
I believe it could be fixed by update sensor keys  fro M2Pro - found and updated my list comment with SMC keys:
https://github.com/exelban/stats/issues/1703

joujek, 2023-12-14:
I found interesting statement mentioned by TG Pro team:

> This was a tricky one since the new models have a slightly different way of exposing the temperatures and they also **do not report valid numbers if the CPU usage is low**. We were able to find a way around this and it will continuously update the temperatures just like every other Mac.

https://www.tunabellysoftware.com/blog/files/m2pro-plus-m2max-temperatures-and-fan-control.html