exelban/stats #1700: Crashes when opening battery popup

bruhtazor, 2023-11-25:
Whenever i open the battery icon and i click on the popup the app quits. 
I'm running the app on a macbook pro late 2015 on MacOS Montery 12.6.9.

exelban, 2023-11-25:
Please share the crash report.

bruhtazor, 2023-11-25:
> Please share the crash report.

Where can i find the crash report ?

exelban, 2023-11-25:
It could be found in Applications -> Utilities -> Console. Find the Crash Reports tab. There you must see a row called Stats.

bruhtazor, 2023-11-25:
> It could be found in Applications -> Utilities -> Console. Find the Crash Reports tab. There you must see a row called Stats.

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [37954]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.9.12 (508)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-11-25 12:52:42.6596 +0100
OS Version:            macOS 12.6.9 (21G726)
Report Version:        12
Anonymous UUID:        E511B9E6-8393-7EA6-3EF9-179A65393503

Sleep/Wake UUID:       6A5D81DB-8403-4280-B333-7DCD2997A6C8

Time Awake Since Boot: 1200000 seconds
Time Since Wake:       864 seconds

System Integrity Protection: enabled

Crashed Thread:        6  Dispatch queue: eu.exelban.Stats.Repeater

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 4 Illegal instruction: 4
Terminating Process:   exc handler [37954]

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff80c5e896a mach_msg_trap + 10
1   libsystem_kernel.dylib        	    0x7ff80c5e8cd8 mach_msg + 56
2   CoreFoundation                	    0x7ff80c6ec34d __CFRunLoopServiceMachPort + 319
3   CoreFoundation                	    0x7ff80c6ea9d8 __CFRunLoopRun + 1276
4   CoreFoundation                	    0x7ff80c6e9e1c CFRunLoopRunSpecific + 562
5   HIToolbox                     	    0x7ff8153975e6 RunCurrentEventLoopInMode + 292
6   HIToolbox                     	    0x7ff81539734a ReceiveNextEventCommon + 594
7   HIToolbox                     	    0x7ff8153970e5 _BlockUntilNextEventMatchingListInModeWithFilter + 70
8   AppKit                        	    0x7ff80f124f6d _DPSNextEvent + 927
9   AppKit                        	    0x7ff80f12362a -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1394
10  AppKit                        	    0x7ff80f115cd9 -[NSApplication run] + 586
11  Stats                         	       0x101051070 0x10104d000 + 16496
12  dyld                          	       0x1017ba52e start + 462

Thread 1:: com.apple.CFSocket.private
0   libsystem_kernel.dylib        	    0x7ff80c5f0d4a __select + 10
1   CoreFoundation                	    0x7ff80c713659 __CFSocketManager + 641
2   libsystem_pthread.dylib       	    0x7ff80c6254e1 _pthread_start + 125
3   libsystem_pthread.dylib       	    0x7ff80c620f6b thread_start + 15

Thread 2:: com.apple.NSURLConnectionLoader
0   libsystem_kernel.dylib        	    0x7ff80c5e896a mach_msg_trap + 10
1   libsystem_kernel.dylib        	    0x7ff80c5e8cd8 mach_msg + 56
2   CoreFoundation                	    0x7ff80c6ec34d __CFRunLoopServiceMachPort + 319
3   CoreFoundation                	    0x7ff80c6ea9d8 __CFRunLoopRun + 1276
4   CoreFoundation                	    0x7ff80c6e9e1c CFRunLoopRunSpecific + 562
5   CFNetwork                     	    0x7ff811313968 0x7ff8110cc000 + 2390376
6   Foundation                    	    0x7ff80d541984 __NSThread__start__ + 1009
7   libsystem_pthread.dylib       	    0x7ff80c6254e1 _pthread_start + 125
8   libsystem_pthread.dylib       	    0x7ff80c620f6b thread_start + 15

Thread 3:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	    0x7ff80c5e896a mach_msg_trap + 10
1   libsystem_kernel.dylib        	    0x7ff80c5e8cd8 mach_msg + 56
[... 347 lines cut ...]
    "queue" : "eu.exelban.Stats.Repeater"
  }
},
  "trialInfo" : {
  "rollouts" : [
    {
      "rolloutId" : "63582c5f8a53461413999550",
      "factorPackIds" : {

      },
      "deploymentId" : 240000002
    },
    {
      "rolloutId" : "5fb4245a1bbfe8005e33a1e1",
      "factorPackIds" : {

      },
      "deploymentId" : 240000021
    }
  ],
  "experiments" : [

  ]
}
}

closed 2023-12-10 by exelban