exelban/stats #1703: SMC sensor keys for Apple M3 and M3 Pro/Max

tycho, 2023-11-27:
**Describe the bug**
I am sure this will be a fun one to figure out, but it would be nice if Stats could recognize and handle the SMC sensors for the Apple M3 (and Pro/Max) chips. The keys are definitely different from M1 and M2. Here's what Stats currently sees (during a CPU-intensive build, so the sensors look a bit alarming):

[image]

It's pretty clear it's not reading the right keys, particularly for CPU P-core/E-core temperature sensors and the GPU.

I've written a little script to try and identify the right sensors, but there is a lot of data making it difficult to correlate. Some of the SMC values are likely not instantaneous values but rather aggregates over certain time windows (e.g. moving averages or similar).

Here's the work-in-progress script. I've added some comments to it to explain the design. I also have a few guesses in there for what some of the keys correlate to.

https://gist.github.com/tycho/73fb64e7ba0b259923a92a4bacea9e78

If you have any tips on identifying the right sensors for temperature, power, etc, I'd appreciate suggestions. 

**Details:**
 - Device: MacBook Pro late 2023, Apple M3 Max
 - macOS: 14.1.1
 - Application version: 2.9.12

exelban, 2023-11-27:
Hi.
Yeap, looks like each new SoC from Apple has different sensor keys.
You can monitor all keys in the Stats by enabling the `Show unknown sensors` option in the settings.

If you are sure the keys you have in the script is the correct I can add them to the Stats.

tycho, 2023-11-27:
The ones I have in the script are currently just guesses. I am somewhat confident the P-core/E-core temperature ones are right, but I'll keep tinkering with this to build some more confidence. I'm pretty sure the GPU-related ones are wrong, I'm still working on hunting those down.

exelban, 2023-11-27:
PS: you have duplicated `Tf0A` and `Tf4A` keys

tycho, 2023-11-27:
> PS: you have duplicated `Tf0A` and `Tf4A` keys

Indeed I did, whoops. The E-core sensors are hard to figure out because of how challenging it is to get a heavy enough user workload to stay running on those (even with "background" QoS class, low niceness, etc). Maybe I'll try triggering a Spotlight rebuild or something, that should put some load on them. And I know that is pretty sticky to E-cores.

tycho, 2023-11-27:
Okay, I updated the script and ran some more workloads to try and figure out the E-core/P-core/GPU temperature sensors.

The P-core ones I'm pretty certain are right.

The E-core ones are probably correct, but it's hard to tell. Even under load (Spotlight reindexing and Time Machine backup), they only get as high as about 54C and idle at 48C. I didn't see any other sensors that were obviously correct though.

As for the GPU ones, they're very definitely *correlated* (rising and dropping based on the GPU workload), but there sure are a lot of them. I mean, yes, there are 40 GPU cores in this thing, but 14 sensors for that seems like too many.

exelban, 2023-11-28:
@tycho fck, it's a pretty nice script)
Thanks for sharing)

Will try to find an m2 platform to get sensors for that soc. Because m2 has a pretty big gap in the sensors.

I want to add this functionality to the app now😅

tycho, 2023-11-29:
I've got an M2 MacBook Air I could tinker with getting some more sensor info from. I'll have to see what I can find.

gilby101, 2023-11-29:
On my M3 (not Pro or Max) my guesses using smc (and comparison with watch-smc and values.swift)  are:

P-Cores: There are no Tf keys - so completely different. I think the 4 cores are:
Tp01 Tp05 Tp0D Tp0H which repeat very closely as Tp0L Tp0P Tp0X Tp0b
This is more like your base Apple silicon keys.

E-Cores: I am happy with Te05 and Te0L (as in your list). I think Te0P is wrong and Te0S does not exist. 
I don't know where the other 2 E-cores might be!

tycho, 2023-11-29:
Interesting! Could you share the output of "/Applications/Stats.app/Contents/Resources/smc list"? It would be interesting to know which keys the M3 and M3 Max have in common -- any added keys in the M3 Max might be the additional cores and could give hints as to the naming scheme.

gilby101, 2023-11-29:
Here you are: [Uploading smc20231129-1527.txt…]()

tycho, 2023-11-29:
> Here you are: Uploading smc20231129-1527.txt…

Try again? It didn't finish uploading before you hit submit. :)

gilby101, 2023-11-29:
Try again:
[smc20231129-1527.txt](https://github.com/exelban/stats/files/13496058/smc20231129-1527.txt)

tycho, 2023-11-29:
> Try again: [smc20231129-1527.txt](https://github.com/exelban/stats/files/13496058/smc20231129-1527.txt)

Wow, that's a lot more differences than I was expecting.
https://gist.github.com/tycho/215aaf4b9eedc065f132cf6d98153d00

gilby101, 2023-11-29:
Very different. Maybe the M3 is just a souped up M1 design and its sensors are more like an M1.  This is my only AS Mac so can't compare.

My interest is that, for my own amusement, I use Cacti (SNMP network monitoring) to capture a few Mac sensors and have always based that on running smcutil/smc under an SNMP script.

tycho, 2023-11-29:
I updated the `watch-smc.py` script, embedding the lists of M3 and M3 Max SMC keys. This way we can easily identify which ones are in common between the two.

For example, here's what happens with the Power-related SMC keys during a GPU-based (MetalRT on) Blender render on my M3 Max:

```
[PC40] M3-Max  :   7.625   L  -0.033 B  +7.625 GPU 16
[PHPC] M3-Any  :  58.209   L -11.954 B +50.133 CPU power total? 1
[PHPS] M3-Any  :  50.055   L  +3.097 B +48.675 CPU power total? 2
[PMVC] M3-Max  :  15.066   L  +1.229 B +13.043
[PPBR] M3-Any  :  76.915   L  +2.905 B +69.903
[PPSC] M3-Max  :   8.872   L  +0.749 B  +8.271
[PSTR] M3-Any  :  75.708   L  +2.332 B +65.885
[PSVR] M3-Max  :  53.357   L  +2.042 B +52.024
[PZC0] M3-Any  :  50.051   L  +3.146 B +48.671
[PZC1] M3-Any  :  49.765   L  +2.555 B +48.402
[PZD1] M3-Any  :   8.413   L  +0.971 B  +8.252
[Pb0f] M3-Any  :  76.989   L  +3.093 B +69.952 Battery discharge rate
```
During this run, `powermetrics --samplers gpu_power` claims `GPU Power: 29607 mW`. Nothing above seems to directly correlate. Hrm.

tycho, 2023-11-29:
I've added some notes on here: https://gist.github.com/tycho/73fb64e7ba0b259923a92a4bacea9e78

To summarize, my current line of thought on the breakdown for the PCxx values is:
```
First 6 P-cores        PC02
Second 6 P-cores       PC42
E-cores                PC32
Generic GPU cores      PC10, PC12, PC20, PC22
Compute GPU cores?     PC40
```
I'm not completely sure what PC40 is, but a Blender render (with **or** without MetalRT) will light that one up.

Unfortunately all the PCxx values are not present on the M3 base model, only on M3 Max. It's entirely possible that M3, M3 Pro, and M3 Max all have different SMC keys.

tycho, 2023-11-29:
This looks promising:

https://github.com/freedomtan/test-ioreport

If you use that, one of the bits in the report it spews out is this (truncated because there's a lot):
```
...
2023-11-29 04:04:33.131 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  PCPU1DTL59: 1226299 (mJ)
2023-11-29 04:04:33.131 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  PCPU1DTL5a: 1165671 (mJ)
2023-11-29 04:04:33.131 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  PCPU1DTL5b: 510189 (mJ)
2023-11-29 04:04:33.131 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  PCPU1DTL5c: 236246 (mJ)
2023-11-29 04:04:33.131 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  PCPU1DTL5d: 204126 (mJ)
2023-11-29 04:04:33.131 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  PCPU1DTL5e: 284341 (mJ)
2023-11-29 04:04:33.131 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  PCPU1DTL5f: 158006 (mJ)
2023-11-29 04:04:33.131 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  PCPU1DTL510: 472478 (mJ)
2023-11-29 04:04:33.131 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  PCPU1DTL511: 397702 (mJ)
2023-11-29 04:04:33.131 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  PCPU1DTL512: 131972 (mJ)
2023-11-29 04:04:33.132 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  PCPU1DTL513: 14117558 (mJ)
2023-11-29 04:04:33.132 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  GPU0: 146981507 (mJ)
2023-11-29 04:04:33.132 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  GPU CS0: 25159023 (mJ)
2023-11-29 04:04:33.132 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  GPU SRAM0: 745944 (mJ)
2023-11-29 04:04:33.132 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  GPU CS SRAM0: 42309 (mJ)
2023-11-29 04:04:33.132 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  ANE0: 1838325 (mJ)
2023-11-29 04:04:33.132 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  ISP0: 1029132 (mJ)
2023-11-29 04:04:33.132 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  AVE0: 1386731 (mJ)
2023-11-29 04:04:33.132 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  MSR0: 1506437 (mJ)
2023-11-29 04:04:33.132 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  DCS0: 88926760 (mJ)
2023-11-29 04:04:33.132 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  DRAM0: 101960769 (mJ)
2023-11-29 04:04:33.132 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  AMCC0: 175035374 (mJ)
2023-11-29 04:04:33.132 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  DISP0: 29543093 (mJ)
2023-11-29 04:04:33.132 test_ioreport[73270:3609034] AppleT6031PMGR <id 0x1000003f9>: Energy Model: (null):  DISPEXT0: 8106668 (mJ)
```
It reports energy in milli-Joule counters. So you can take two snapshots, subtract to get deltas between the two snapshots, divide by time between reading samples, and ta-daa, you have watts for each metric.

That seems to be the interface that the `powermetrics` command line tool talks to.

@exelban Think you could try out integrating an IOReport method for gathering power metrics?

tycho, 2023-12-03:
[image]

@exelban The sensor list is still a bit wonky.

exelban, 2023-12-03:
oh, I forgot to detect m3 platform(
will deploy a quick fix in a few minutes

exelban, 2023-12-03:
Could you please check this build:

[Stats.dmg.zip](https://github.com/exelban/stats/files/13539246/Stats.dmg.zip)

djcristi, 2023-12-09:
the cpu sensor temps are not ok. they show like 10 degree less than powermetrics / istat / tg pro . on my 16" m3 max 14 core.

djcristi, 2023-12-09:
PDBR i suspect it's Display. i see it goes from 1W to 11W based on backlight.

tycho, 2023-12-10:
> PDBR i suspect it's Display. i see it goes from 1W to 11W based on backlight.

Agreed, already had that one here:
https://gist.github.com/tycho/73fb64e7ba0b259923a92a4bacea9e78#file-watch-smc-py-L620
DBR probably stands for "Display BRightness"

tycho, 2023-12-10:
> the cpu sensor temps are not ok. they show like 10 degree less than powermetrics / istat / tg pro . on my 16" m3 max 14 core. also it displays 12 p-cores temp values and it only has 10

Can you post your "smc list" output as well? That might help narrow down the core temp sensor possibilities.

djcristi, 2023-12-10:
[smc.txt](https://github.com/exelban/stats/files/13626994/smc.txt)
[image]

tycho, 2023-12-10:
@djcristi Updated with your 14-core M3 Max keys:

https://gist.github.com/tycho/73fb64e7ba0b259923a92a4bacea9e78

I added a quick way to see the difference between the 14 and 16 core versions:

> Keys in 16-core M3 Max but not in 14-core M3 Max:
> ['SFFC', 'TDTC', 'TVS3', 'Tf2C', 'ceF3', 'ceP3', 'ceU3', 'cm01', 'cmD5', 'cmF3', 'csF3', 'csP3', 'csU3', 'f009', 'f00a', 'frMX', 'fts3', 'fuP3', 'fuU3', 'iaD5', 'iaFs', 'iaP3', 'isF3', 'isP3', 'isU3', 'maF3', 'maP3', 'maU3', 'mlT3', 'mlTL', 'mlTl', 'mm00', 'mmV5', 'mxS3', 'mxSd', 'oF30', 'oF31', 'oF32', 'oF33', 'oF34', 'oF35', 'oF36', 'oF37', 'oF38', 'oF39', 'oF40', 'oF41', 'oF42', 'oTO3', 'ocew', 'ofci', 'ofcj', 'ofck', 'ofcl', 'ofli', 'oflj', 'oflk', 'ofll', 'ofp3', 'oft2', 'oft4', 'oft6', 'oftA', 'oftc', 'ofte', 'oisw', 'oke8', 'oki4', 'oki8', 'okiw', 'okp4', 'okp8', 'okpw', 'otCZ', 'otCz', 'otMx', 'otMy', 'ov04', 'ov05', 'ov24', 'ov40', 'ov41', 'ov42', 'ov43', 'ov44', 'ovCY', 'ovCj', 'ovCk', 'ovCl', 'ovEa', 'ovEb', 'ovEc', 'ovEg', 'ovNj', 'ovNk', 'ovNl', 'si0J', 'si0T', 'si0e', 'si0f', 'si0g', 'sit5', 'uuF3', 'uuO3', 'uuP3', 'uuU3', 'uuV5', 'uuhp', 'w002']

So there are a few extra temperature sensors in there, but they're kind of far apart naming-wise. No extra power sensors.

djcristi, 2023-12-10:
did you notice the cpu power consumption diff? so in my 13" pro m1 , the powermetrics power is the same with the one reported by the one from smc. In this m3 max 14c , under full p-cores load smc reports ~60W (62 phpc/ 58 phps, 75 pstr , 72 pb0f) and powermetrics only 48W.
on low power mode:  smc reports ~30W (32 phpc/ 30 phps, 46 pstr , 44 pb0f) and powermetrics only 21W.

Considering total power consumption, i guess that powermetrics is wrong.

joujek, 2023-12-11:
Hi @exelban 
I would like ask again for update sensor keys for M2 Pro / Max platform to have Efficiency and Performance cores instead of 8 general:

[image]

[image]

[image]

exelban, 2023-12-12:
@joujek no problem. Just give keys that need to be used instead of actual.

joujek, 2023-12-12:
thats the problem I am not sure which keys are correct once - Anyone can help?
I will try to figure out and update this thread

joujek, 2023-12-13:
In addition I'm attaching dump for M3Pro Max Platform: Mac15,3, Mac15,6, Mac15,8, Mac15,10, Mac15,7, Mac15,9, Mac15,11

[M3ProMax_dump.txt](https://github.com/exelban/stats/files/13662986/M3ProMax_dump.txt)

gilby101, 2023-12-14:
@joujek 
I have an M3. I agree (from observation of counters under loads) with your "choice" for GPU and P-core sensors.

Your E-cores, I am not so sure about.  When P-cores are loaded and the E-cores not, your values for the E-cores are roughly the same as the P-cores - I would expect them to be lower.

My best guess is Te05 and Te0L for two E-cores, but I am stuck for the other two! All the other Te values are higher than those.

Looking at other products: TGPro shows e-cores lower than p-cores and agree with using Te05 and Te0L. iStat shows e-cores always about the same as p-cores - suggesting it uses Tp0B, etc for E-cores.

joujek, 2023-12-14:
HI @gilby101   - The dump I shared is from latest iStat  ;)

Personally I have M2 Pro 12/19  and compared iStats vs Macs Fan Control vs Stats (unknown sensors list).
I believe for M2Pro/Max any changes will be better than generic 8 cores we have now in Stats.

gilby101, 2023-12-14:
Hi @joujek  My feeling is that TG Pro is ahead of iStats for the M3 (not pro/max). I will leave you do any serious reverse engineering.

djcristi, 2023-12-16:
macbook pro 14" m3 max 14c 
[image]
[smclist.txt](https://github.com/exelban/stats/files/13692705/smclist.txt)

djcristi, 2023-12-16:
> did you notice the cpu power consumption diff? so in my 13" pro m1 , the powermetrics power is the same with the one reported by the one from smc.
> ----- blabla
> Considering total power consumption, powermetrics is wrong.
> 
> @tycho can you test on your 16 core to see how this looks?

as i've wrote on my previous post, i did more findings today.

same test: https://cpux.net/cpu-stress-test-online , 10 thread at 100%

14” m3 max 14c

Low power
	freq : p-core 2.6 idle/ 2.17 load ; e-code 1-2 idle/ 2.55 load 
	load: 
		power: smc reports 26W phpc (9.5 display pdbr, 42 total system), and powermetrics only 15W cpu/pkg.
		performance: 1300 h/s

[image]

Auto/high-power
	freq : p-core 2.7-4 idle/ 3.5 at start load ; e-code 1-1.3 idle/ 2.4 load 
	load: 
		power: smc reports 62W phpc (9.5 display pdbr, 85 total system), and powermetrics only 46W cpu/pkg.
		performance: 2100 h/s
	
	but as you can see in the graph after short time it stats throttling, and fans slowly increase
	after I maxed the fans the power/perf increased, keeping a consistent on this values:

	freq : p-core  3.2
	load: 
		power: smc reports 50W phpc (9.5 display pdbr, 75 total system), and powermetrics only 35W cpu/pkg.
		performance: 1900 h/s

[image]

===================

16” m3 max 14c

Low power
	freq : p-core 2.62 idle/ 2.5 load ; e-code 1-2 idle/ 2.55 load 
	load: 
		power: smc reports 33W phpc (12w display pdbr, 52 total system), and powermetrics only 21W cpu/pkg.
		performance: 1600 h/s

Auto/high-power
	freq : p-core 3.5-3.7 idle/ 3.58 load ; e-code 1-1.3 idle/ 2.55 load 
	load: 
		power: smc reports 60W phpc (12w display pdbr, 84 total system), and powermetrics only 48W cpu/pkg.
		performance: 2120 h/s

	does not throttle at all, at some point I’ve set the fans to maximum but that decreased temps , so the 16 has a lot more case+fans+heatpipe cooling power

joujek, 2023-12-18:
> UPDATE:
> 
> HI @exelban below the keys found by decompiling one of alternative sw: I have compared to my smc output and values from "Show unknown sensors" looks legit :)
> 
> Px - x performance core / Ex - x efficiency core
> 
> M2 8C: Tp01 P1 Tp05 P2 Tp09 P3 Tp0D P4
> 
> Tp0b E1 Tp0f E2 Tp0j E3 **Tp0n** E4 //This sensor is different than Pro/Max platform
> 
> M2 PRO 10C
> 
> Tp01 P1 Tp05 P2 Tp09 P3 Tp0D P4 Tp0X P5 Tp0b P6
> 
> Tp0f E1 Tp1l E2 Tp1h E3 Tp0j E4
> 
> M2 PRO 14C
> 
> Tp01 P1 Tp05 P2 Tp09 P3 Tp0D P4 Tp0X P5 Tp0b P6 Tp0f P7 Tp0j P8
> 
> Tp1h E1 Tp1t E2 Tp1p E3 Tp1l E4
> 
> ```
> case .m2, .m2Pro, .m2Max, .m2Ultra:
> self.list = ["Tp1h", "Tp1t", "Tp1p", "Tp1l", "Tp01", "Tp05", "Tp09", "Tp0D","Tp0X","Tp0b","Tp0f","Tp0j"]
> ```
> 
>  ```
> //M2
>     Sensor(key: "Tp1h", name: "CPU efficiency core 1", group: .CPU, type: .temperature, platforms: [.m2, .m2Max, .m2Pro, .m2Ultra], average: true),
>     Sensor(key: "Tp1t", name: "CPU efficiency core 2", group: .CPU, type: .temperature, platforms: [.m2, .m2Max, .m2Pro, .m2Ultra], average: true),
>     Sensor(key: "Tp1p", name: "CPU efficiency core 3", group: .CPU, type: .temperature, platforms: [.m2, .m2Max, .m2Pro, .m2Ultra], average: true),
>     Sensor(key: "Tp1l", name: "CPU efficiency core 4", group: .CPU, type: .temperature, platforms: [.m2, .m2Max, .m2Pro, .m2Ultra], average: true),
>     
>     Sensor(key: "Tp01", name: "CPU performance core 1", group: .CPU, type: .temperature, platforms: [.m2, .m2Max, .m2Pro, .m2Ultra], average: true),
>     Sensor(key: "Tp05", name: "CPU performance core 2", group: .CPU, type: .temperature, platforms: [.m2, .m2Max, .m2Pro, .m2Ultra], average: true),
>     Sensor(key: "Tp09", name: "CPU performance core 3", group: .CPU, type: .temperature, platforms: [.m2, .m2Max, .m2Pro, .m2Ultra], average: true),
>     Sensor(key: "Tp0D", name: "CPU performance core 4", group: .CPU, type: .temperature, platforms: [.m2, .m2Max, .m2Pro, .m2Ultra], average: true),
>     Sensor(key: "Tp0X", name: "CPU performance core 5", group: .CPU, type: .temperature, platforms: [.m2, .m2Max, .m2Pro, .m2Ultra], average: true),
>     Sensor(key: "Tp0b", name: "CPU performance core 6", group: .CPU, type: .temperature, platforms: [.m2, .m2Max, .m2Pro, .m2Ultra], average: true),
>     Sensor(key: "Tp0f", name: "CPU performance core 7", group: .CPU, type: .temperature, platforms: [.m2, .m2Max, .m2Pro, .m2Ultra], average: true),
>     Sensor(key: "Tp0j", name: "CPU performance core 8", group: .CPU, type: .temperature, platforms: [.m2, .m2Max, .m2Pro, .m2Ultra], average: true),
> ```
> 
> Thank you in advance

@exelban any feedback on this ?

closed 2023-12-31 by exelban

djcristi, 2024-01-13:
@exelban do you have on plan to fix temp sensors for m3 max, it does not show correct temp , i guess you've messed the sensor names. 

also the PDBR (mini-led backlight power) i think can be put as known sensor, it's common along Macbook xdr display (14/16 mbp)

exelban, 2024-01-13:
@djcristi all sensors from that issue have already been included in the build for a few weeks.
If you have more sensors just share the list and I will add it.

djcristi, 2024-01-14:
like i've wrote in my previous thread, PDBR is still listed as unknown istead of 'Display Backlight Power' or something.

and the temp sensor names are wrong , for M3 max 14c at least, 

stats (wrong <40)
[image]

tg pro (real ones):
[image]

istat menus (wrong on 9,10):
[image]

xdsta, 2024-01-14:
Just updated to version Version 2.9.20:

[image]

[M3ProMax_dump.txt](https://github.com/exelban/stats/files/13931581/M3ProMax_dump.txt)

Seems like the values from commit https://github.com/exelban/stats/commit/df1a0a8bacb9a9a6c23afa3c5faaabae2fc15890 are not working for M3 pro 11C CPU.